bblanchon/ArduinoJson #810: Don't Work Parser (ESP8285 Arduino)

mcmega, 2018-09-14:
Hi, Benoît Blanchon!
Found an error in the library.
I am use ESPAsyncWebServer with Web Socket. Packages arrive, but are not processed on the new version.
Version 6.3.0 it Work.
Version 6.4.0 is Don't Work.
```cpp
void onEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t * data, size_t len) {
  if (type == WS_EVT_CONNECT) {
    client->ping();
  } else if (type == WS_EVT_DATA) {
    DynamicJsonDocument doc;
    DeserializationError error = deserializeJson(doc, (char*)data);
    if (error) {
      sendUART("ER!", "WS-Data", "Parser");
      return;
    }
    JsonObject root = doc.as<JsonObject>();
    if (root["actualInfo"]) { //Handler actualInfo }
    else if (root["settings"]) { //Handler settings }
    //...
    else { //Handler Unknown Data }
  }
}
```

bblanchon, 2018-09-14:
Hi @mcmega,

Thanks for reporting this problem.
However, I don't have enough information to fix this bug.

What is the value of `error`?
What is the JSON input exactly?
Can you create an [MCVE](https://stackoverflow.com/help/mcve) that doesn't involve ESPAsyncWebServer?

Arduino 6.4 uses slightly more RAM than the previous versions.
Is it possible that you're running out of heap memory?
Can you try to specify the capacity to the constructor of `DynamicJsonDocument`?
The default is 256, please try 512 and 1024.

Regards,
Benoit

mcmega, 2018-09-14:
```error``` value is 0
JSON package is correct ```(char*)data```
Version 6.3.0 it work. There were no problems.
Version 6.4.0 why does not it parse the package and ```if (root["..."])``` do not work.

mcmega, 2018-09-14:
I have a lot of key checks, how do I do it right?
```cpp
JsonVariant key1= root["key1"];
if (!key1.isNull()) {
  //...
}
//... more 25 keys
JsonVariant keyN= root["keyN"];
if (!keyN.isNull()) {
  //...
}
```

bblanchon, 2018-09-17:
I you have a lot of keys, it's better to iterate:

```c++
for(JsonPair p : root) {
   if (!strcmp(p.key(), "key1")
      handleKey1(p.value());

   // ...
}
```

As you can see, there is no lookup and no need to check for null.

@mcmega, is the initial problem solved?

mcmega, 2018-09-17:
I conditionally called the key1 .. keyN. 
The keys are all different. Will this also work in this case?

bblanchon, 2018-09-19:
Hi @mcmega,

You can see a `JsonObject` as a collection of `JsonPair`.
A `JsonPair` contains a key and a value.
So, yes, `p.key()` is the key and `p.value()` is the value.

If you expect one of the values to be an array, change the type of the argument:

```c++
void handleN(JsonArray array) {
   //  ...
}
```

The `JsonVariant` returned by `p.value()` will by casted implicitly to `JsonArray`.

BTW, is the initial problem solved?

Regards,
Benoit

mcmega, 2018-09-24:
Yes, all is well!
Thank you! Good library!

closed 2018-09-24 by mcmega