bblanchon/ArduinoJson #812: keys in PROGMEM?

MarcFinns, 2018-09-15:
Question. In my effort to save memory i stored strings in progmem as below.

const char PROGMEM google_key[]  = "google_key";

The following sequence crashes in the last line. Is this a legal way to achieve the memory saving...? thanks!
 
       configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject &json = jsonBuffer.parseObject(buf.get());

        if (json.success())
        {
          **strcpy(config.mqtt_server, json[google_key]);**

bblanchon, 2018-09-16:
Hi @MarcFinns,

As with many other Arduino libraries, `PROGMEM` strings must have the type `const __FlashStringHelper*` to be detected as such.
You can simply cast the pointer, but the easiest way is to use the `F()` macro:

```c++
strcpy(config.mqtt_server, json[F("google_key")]);
```

Also, to prevent the undefined behavior of `strcpy()` when one argument is `NULL`, I recommend that you use the [operator `|`](https://arduinojson.org/v5/api/jsonvariant/or/?utm_source=github&utm_medium=issues):

```c++
strcpy(config.mqtt_server, json[F("google_key")] | "default");
```

Moreover, you can protect your program against buffer overruns by using `strlcpy()` instead of `strcpy()`:

```c++
strlcpy(config.mqtt_server, json[F("google_key")] | "default", sizeof(config.mqtt_server));
```

Regards,
Benoit

MarcFinns, 2018-09-16:
Hi Benoit, thanks a million for taking the time.
Indeed i could use the F() macro, but it does not pool strings.. in any case, thanks fr the explanation and keep up the excellent work.

bblanchon, 2018-09-17:
Hi @MarcFinns,

I was so surprised with what you just said; I had to see that by myself!

Indeed, `F()` doesn't perform [string interning](https://en.wikipedia.org/wiki/String_interning).
It's completely obvious when you think about it, but it's counter-intuitive because you expect the macro to behave like a classic string literal.
Thank you very much for pointing that out 👍 

I added the information on the FAQ: [Why does my device crash or reboot?](https://arduinojson.org/v5/faq/why-does-my-device-crash-or-reboot/?utm_source=github&utm_medium=issues).

Regards,
Benoit

closed 2018-09-17 by bblanchon