bblanchon/ArduinoJson #813: Invalid json output after many valid passes

cfeied, 2018-09-16:
I am using ArduinoJson 5.11.1 on a Teensy3.6 with code compiled under VisualMicro / VisualStudio 2017. 

I have encountered a scenario in which AJ produces correct serialized output for several thousand calls, and then when called with the exact same input it suddenly produced two empty (undefined) values in a json string.  This naturally caused an "invalid json string" exception on the receiving host.

My project has been stable and the json strings have been valid for many months. So far as I can detect, no changes were made to any of the relevant classes nor to the serialization code below.  

However, yesterday I did add AJ #defines to allow me to send values like epoch seconds and epoch milliseconds (as longlong or double) without truncation.  
```
#define ARDUINOJSON_USE_DOUBLE 1
#define ARDUINOJSON_USE_LONG_LONG 1
```
After adding those defines, the system ran correctly for many hours yesterday. Today after about an hour (2356 valid json strings generated by this exact code path) I suddenly encountered the invalid json string exception.  Here is the serialization code:

```
void AddNewRespEvent(EventType type,EventSubType subtype, char* thisEvent, char* notes)
{
	StaticJsonBuffer<2000> jsonBuffer;
	JsonObject& RespEvent_Json = jsonBuffer.createObject();
	RespEvent_Json["ID"] = GetNextEventID();
	RespEvent_Json["DateTimeLocal"] = SnapTime.DT_Local_MillisCharFmt;
	RespEvent_Json["DateTimeUTC"] = SnapTime.DT_UTC_MillisCharFmt;
	RespEvent_Json["SessionTime"] = SnapTime.SessionMillisCharFmt;
	RespEvent_Json["Type"] = type.ToString();  
	RespEvent_Json["SubType"] = subtype.ToString();
	RespEvent_Json["Event"] = thisEvent;
	RespEvent_Json["Notes"] = notes;

	char event[500];
	int bytesPrinted = RespEvent_Json.printTo(event);	
        SendToHost(event);
        // BUG: after an hour of perfectly formed output (2356 valid strings) 
        // the Host received a single malformed JSON string that had no value 
       // for Type or Subtype. Both the values and the surrounding quotes were missing.
}
```
These calls are made from a single method and the EventType and EventSubType are hard-coded for this type of call, so cannot change.  The fact that their String representation is used is potentially problematic, but the Teensy 3.6 has a lot of headroom and my instrumentation shows no likelihood of heap collision due to String duplication. (I'll change this to use a const char* to remove that possibility).

Here is what this output string normally looks like:

`{"ID":2350,"DateTimeLocal":"2018.9.16_15:20:37.500","DateTimeUTC":"2018.9.16_19:20:37.500","SessionTime":"00:59:23.500","Type":"Stimulus","SubType":"Unknown","Event":"Stimulus (Off) \n\tStimulus OFF in setting","Notes":""}`

And here is the invalid json string output that caused the exception:

`{"ID":2357,"DateTimeLocal":"2018.9.16_15:20:41.504","DateTimeUTC":"2018.9.16_19:20:41.504","SessionTime":"00:59:27.504","Type":,"SubType":,"Event":"Stimulus (Off) \n\tStimulus OFF in setting","Notes":""}`

You can see that there are no values given for Type or SubType, and of course this json fails validation everywhere.  

Is this an issue that has been seen before?  Any clues or suggestions?  

This is a small string and I could easily encode it manually, but I also serialize a number of much larger objects using iterators, and manual encoding would be a real burden. 

Thanks!

cfeied, 2018-09-17:
On the second test pass the very same error occurred after 2 hrs and 25 minutes (13,179 good json strings followed by a bad string).  This time three elements are null: one of them is a quoted json null and the other two are invalid json, as before.  The data passed to ArduinoJson is hardcoded and none of the three elements have any code path that could make them null, but even if the inputs were null it seems hard to believe that AJ should be emitting invalid JSON primitives.

The 13,179 valid json strings look just fine. Here's string  #13180 as output from ArduinoJson in this case:

'{"ID":13180,"DateTimeLocal":"2018.9.17_22:19:40.684","DateTimeUTC":"2018.9.17_02:19:40.684","SessionTime":"02:25:45.684","Type":,"SubType":,"Event":"(null)","Notes":""}'

For test #3 I'm going to comment out the AJ `#define` statements (and comment out the epoch seconds and epoch millis, which have been working just fine) and see whether it makes any difference.

bblanchon, 2018-09-17:
Hi @cfeied,

There is one thing that immediately strikes me: the size of the variables in the stack.
I'm not sure how the ARM processor deals with stack overflows, but it could certainly explain the strange behavior.
Can you try to switch to a [`DynamicJsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues)?

Also, if you use several compilation units (ie several `.cpp` files), make sure that all files define the same values for [ARDUINOJSON_USE_DOUBLE](https://arduinojson.org/v5/api/config/use_double/?utm_source=github&utm_medium=issues) and [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v5/api/config/use_long_long/?utm_source=github&utm_medium=issues).

Regards,
Benoit

cfeied, 2018-09-17:
Thank you, Benoit. 

I changed my code to pass a c_str() from classes EventType and EventSubType, and the behavior changed. This time I received 13,209 valid json strings with valid data, and then AJ produced a valid json string with explicit nulls for those elements.  

```
	RespEvent_Json["Type"] = type.c_str();  
	RespEvent_Json["SubType"] = subtype.c_str();
```

`{"ID":13210,"DateTimeLocal":"2018.9.17_06:06:41.342","DateTimeUTC":"2018.9.17_10:06:41.342","SessionTime":"02:26:31.342","Type":null,"SubType":null,"Event":"\n\tPosition: Prone","Notes":""}`

That seems to indicate that I am facing two possibly separate bugs here. The first is a bug in my program that causes some of my validated const String fields to become suddenly null.

The second is a bug in ArduinoJson that (in my program context) produces a valid json string when passed a null const char*, but produces an invalid json string when passed a null String. 

I'll now attempt to repro this behavior in a minimal program to see whether ArduinoJson behaves the same way when passed null Strings in an otherwise normally-functioning program, or whether the same underlying problem is affecting both my code and the behavior of ArduinoJson.

I did check all compilation units and my ArduinoJson #defines appear in only one place.  Also the behavior is the same without the #defines. 

My instrumentation shows plenty of stack space, and in fact in other parts of the code I use staticJsonBuffer<8000> with no problems -- but I agree that weird String behavior points that way, and I'll try again switching to a dynamic buffer (in the past I saw data truncation with dynamic buffers).  

Once I resolve this issue I'll also go through and purge my classes of String types completely.

bblanchon, 2018-09-18:
I handled the case of null `String` in ArduinoJson; however, it's a feature that is not well covered in the unit tests because `std::string` never returns null.

Anyway, I see that you know exactly what you're doing.
Please inform me of your program and tell me if I can be of any help.

cfeied, 2018-09-22:
I have found my issue: a glitch in another library trashed some Strings in my heap, and ArduinoJson was fed Strings that were null.  

Even though this should not happen, **I believe it would be preferable if AJ did not produce invalid json as an output**, for obvious reasons.  

The desired behavior would be to treat null Strings in exactly the same way that a null char* is treated: the output should be `null` rather than merely empty.

Thank you very much for your help. 

BTW, I greatly enjoyed your book and found your explanations both clear and elegant. Your writing reminds me of John Skeet's books, which always manage to cut through the complexity in a natural and intuitive manner.

bblanchon, 2018-09-23:
Hi @cfeied,

I agree. It should print null in this situation, as version 6 does.

Thank you very much for purchasing the book.
I don't believe that I deserved to be compared with The Skeet, but thanks :-)

Regards,
Benoit

bblanchon, 2018-09-23:
Hi again!

I didn't manage to reproduce the bug.
I started from `JsonGeneratorExample.ino` and changed the following line:

```c++
root["sensor"] = String((const char*)0);
```

But the output is correct:

```json
{"sensor":null,"time":1351824120,"data":[48.75608,2.302038]}
```

That was on an Arduino Duemilanove, but I don't think it matters.

Can you write an MCVE?

Regards,
Benoit

bblanchon, 2018-10-04:
@cfeied, do you think you can provide an [MCVE](https://stackoverflow.com/help/mcve).
If you cannot, I'll have to close this issue.