bblanchon/ArduinoJson #814: Max number of elements in a JsonArray

gitandlucsil, 2018-09-17:
Hi, I'm using the library in a ESP-01 model using a Arduino IDE.

I'm declaring my buffer and array this way:

 DynamicJsonBuffer jsonBuffer;
 JsonArray& jsonArray = jsonBuffer.createArray();

But I realized that the maximum number of elements stored in the jsonArray (using jsonArray.size() method) is the 128. Is the any way to store more than this or this value is the limit?

Thanks for any help

bblanchon, 2018-09-18:
Hi @gitandlucsil,

ArduinoJson doesn't put any limit on the number of elements in a [`JsonArray`](https://arduinojson.org/v5/api/jsonarray/?utm_source=github&utm_medium=issues).
It seems that your program is running out of memory.
The [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/?utm_source=github&utm_medium=issues) will tell you how much RAM you need to store your array.

See also:

* [How to reduce memory usage?](https://arduinojson.org/v5/faq/how-to-reduce-memory-usage/?utm_source=github&utm_medium=issues)

Regards,
Benoit

gitandlucsil, 2018-09-21:
Thank you, I finally solve my problem.

My trouble was the usage of the JsonObject. I have a siwtch-case inside a while loop, and I was instantiating the JsonObject in any momento inside the switch-case, for example:

switch (index_data) {
JsonObject& jsonObj = jsonBuffer.createObject();
}

So, in any interaction with switch-case, a new instance of jsonObj was create. The solution was just make the instance in the last condtion of the switch-case (that works like a state machine), for example:

case 20:
JsonObject& jsonObj = jsonBuffer.createObject();
jsonObj["filed_name"] = var;
break;

bblanchon, 2018-09-23:
You're welcome, I'm glad you found the problem.
Thank you for using ArduinoJson

closed 2018-09-23 by bblanchon