bblanchon/ArduinoJson #816: no matching function for call to 'println(int*&)

cyborgium, 2018-09-18:
If I compile the below code, I keep getting the following error message:
`no matching function for call to 'println(int*&)'

java.io.IOException: jssc.SerialPortException: Port name - COM4; Method name - setEventsMask(); Exception type - Can't set mask.
	at processing.app.Serial.dispose(Serial.java:166)
	at processing.app.SerialMonitor.close(SerialMonitor.java:116)
	at processing.app.AbstractMonitor.suspend(AbstractMonitor.java:90)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2020)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM4; Method name - setEventsMask(); Exception type - Can't set mask.
	at jssc.SerialPort.setEventsMask(SerialPort.java:279)
	at jssc.SerialPort.removeEventListener(SerialPort.java:1064)
	at jssc.SerialPort.closePort(SerialPort.java:1090)
	at processing.app.Serial.dispose(Serial.java:163)
	... 4 more`

I don't understand what it is that I'm doing wrong. Does anyone see what I need to change?

`#include <FS.h>
#include <SPIFFS.h>
#include <ArduinoJson.h>

 
char * file = "/adc.json";   //File name
 
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
 
  //Initialize File System
  SPIFFS.begin();
  Serial.println("File System Initialized");
 
  File dataFile = SPIFFS.open(file, "r");                 //Open File for reading
  Serial.println("Reading Data from File:");
  //Data from file
  //for(int i=0;i<dataFile.size();i++)                    //Read upto complete file size
  //{
    //Serial.print((char)dataFile.read());                //Read file
  //}
  dataFile.close();
}

void handleReceivedMessage(String message){
 
  StaticJsonBuffer<500> jsondocument;                     //Memory pool
  JsonObject& parsed = jsondocument.parseObject(file);    //Parse message
 
  if (!parsed.success()) {   //Check for errors in parsing
    Serial.println("Parsing failed");
    return;
  }
 
  int * one = parsed["-9.86"];
  int * two = parsed["-10.36"];
  int * three = parsed["-14.86"];
  int * four = parsed["-3.88"];
 
  Serial.print("-9.86: ");
  Serial.println(one);
 
  Serial.print("-10.36: ");
  Serial.println(two);
 
  Serial.print("-14.86: ");
  Serial.println(three);
 
  Serial.print("-3.88: ");
  Serial.println(four);
 
  Serial.println("------------------------------");
}

void loop() {
}`

P.S.: I know I'm on an older version of ArduinoJson

bblanchon, 2018-09-19:
Hi @cyborgium,

Here is a very strange program.

What are you trying to do with the following line?

```c++
int * one = parsed["-9.86"];
```

From what I understand, you're trying to extract an integer from a value named "-9.86", but you mistakenly added a `*`. You need to remove the star because it's not a pointer that ArduinoJson returns, it's an `int`:

```c++
int one = parsed["-9.86"];
```

But, still, it's very curious to have a value named "-9.86", it would mean that the JSON input looks like:

```json
{
   "-9.86": 1,
   "-10.36": 2,
   "-14.86": 3,
   "-3.88": 4
}
```

Although it's perfectly legal JSON, I find it rather strange.

One last thing, you can use the [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/?utm_source=github&utm_medium=issues) to write the program for you.

Regards,
Benoit

cyborgium, 2018-09-19:
Hi,

Yes you were right, thanks for that.
The code now compiles fine, only it doesn't actually parse. Because of the below code, it just prints "Parsing failed" and stops the program cause of the return. I've already tried removing the return to see if maybe it does actually parse, but then it just prints "0" for every value that it is supposed to parse.

```
  if (!parsed.success()) {   //Check for errors in parsing
    Serial.println("Parsing failed");
    return;
  }
```

Any idea to what I'm doing wrong now?

As to what you mentioned about it being an interesting JSON file, I know. That example you gave is exactly what the file looks like.

Thanks

cyborgium, 2018-09-19:
Alright so I managed to make some progress, though it's still not exactly doing what I want it to. See the below code:

```
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  const size_t bufferSize = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3) + 1200;
  DynamicJsonBuffer jsonBuffer(bufferSize);

  //const char* json = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.75608,2.302038]}";
  const char* json = "/adc.json";

  JsonObject& root = jsonBuffer.parseObject(json);

  const char* sensor = root["sensor"]; // "gps"
  long time = root["time"]; // 1351824120

  float data0 = root["data"][0]; // 48.75608
  float data1 = root["data"][1]; // 2.302038

  Serial.println("Data uit json");
  Serial.println(data0);
  Serial.println(data1);
```
The code works fine with the commented out "json" variable, but it doesn't when reading the file. I should really have tried this before....
Does your library support reading from a file? And if not, do you know a way to put the content of the file into a string like that commented out one?

Thanks again.

bblanchon, 2018-09-20:
Hi @cyborgium,

Please see the example [`JsonConfigFile.ino`](https://arduinojson.org/v5/example/config/?utm_source=github&utm_medium=issues); it shows how to read a file from an SD card.

Regards,
Benoit

cyborgium, 2018-09-20:
I've almost got it the way I want it to. The only thing is it that I want to be able to have a variable determine which data it'll get out of the json object. 

`int data0 = root["osp1"][variable];`

I can't seem to get it to work, no matter what datatype I try, it'll always just return 0. How would I do this?

cyborgium, 2018-09-20:
I figured it out myself.

Thanks anyways though!

closed 2018-09-21 by cyborgium

bblanchon, 2018-09-21:
You're welcome @cyborgium; I'm glad you found the problem.
Thanks for using ArduinoJson!