bblanchon/ArduinoJson #817: Buffer was not declared in this scope

KristianDukov, 2018-09-23:
Hello All!

I was trying to fix some code today and I got the following error:

* Compiler error: 

Arduino: 1.8.5 (Windows 10), Board: "WeMos D1(Retired), 80 MHz, 115200, 4M (1M SPIFFS)"

181: error: 'buffer' was not declared in this scope
 root.printTo(buffer);
exit status 1
'buffer' was not declared in this scope

Here is some relevant context:
```
const size_t bufferSize = JSON_OBJECT_SIZE(7); //JSON
DynamicJsonBuffer jsonBuffer(bufferSize); //JSON

JsonObject& root = jsonBuffer.createObject();
JsonObject& root2 = jsonBuffer.createObject();
```
Inside the void loop

```
char buffer[256];
root["reps"] = n;
root["power"] = powerContract;
root["weight"] = weight;
root["set"] = s;
root["contract"] = contract;
root["account"] = "1231234";
root["machine"] = machine;
root.printTo(buffer);
String Json = String(buffer);
```
```
char buffer2[256];
root2["extension"] = xtend;
root["extension"] = xtend;
root2.printTo(buffer2);
root.printTo(buffer);
String Json = String(buffer2);
   //Serial.println(JSONtxt2);  //Debug
  webSocket.broadcastTXT(buffer2);
```

and Finally

```
const char* jsonDuplicate = jsonBuffer.strdup(Json);
array.add(jsonBuffer.parseObject(jsonDuplicate));
array.printTo(Serial);
```

What I am trying to do is create the first JSON and send it via Websocket, get the second one which I also will send via websocket but in the same time add it to the first one, later on added into the array.

bblanchon, 2018-09-24:
Hi @KristianDukov,

```
error: 'StaticJsonBuffer' was not declared in this scope
error: 'DynamicJsonBuffer' was not declared in this scope
error: 'StaticJsonBuffer' does not name a type
error: 'DynamicJsonBuffer' does not name a type
```

These errors usually occur when you compile a program written for ArduinoJson 5, but version 6 is installed on the computer.

Version 6 is currently in beta stage, but unfortunately, the Arduino Library Manager installs it by default. To fix this problem, open the Arduino Library Manager, and downgrade ArduinoJson to 5.13.2.

BTW @KristianDukov, you don't need the variables `buffer` and `buffer2` because you can pass a `String` directly to [`JsonObject::printTo()`](https://arduinojson.org/v5/api/jsonobject/printto/?utm_source=github&utm_medium=issues)

Regards,
Benoit

KristianDukov, 2018-09-25:
Hi @bblanchon 

Thank you for your answer. Unfortunately, the version that I was already using was 5.13.2. I tried to clear the code. Regarding the buffer, if I want to send a websocket message, I still should pass it to buffer, correct?

Regards,
Kris

closed 2018-09-25 by KristianDukov

bblanchon, 2018-09-27:
Hi Kris,

The easiest way to add a nested object in an array is to call [JsonArray::createNestedObject()](https://arduinojson.org/v5/api/jsonarray/createnestedobject/?utm_source=github&utm_medium=issues):

```c++
JsonArray& arr = jsonBuffer.createArray();
JsonObject& merged = arr.createNestedObject();
// now, set the values in the object
arr.printTo(bufferArray);
```

Does that answer your question?

Regards,
Benoit

KristianDukov, 2018-09-27:
Hi Benoit,

The question is exactly in the commented part that you have. Is there a way to add the `createNestedObject()` from the two existing objects `root `and `root2` instead of assigning them `merged["hello"] = "world";`

Thank you for your patience!
Kris

bblanchon, 2018-09-27:
Hi Kris,

You can add `root` and `root2` to the array, like that:

```c++
JsonArray& arr = jsonBuffer.createArray();
arr.add(root);
arr.add(root2);
arr.printTo(bufferArray);
```

Is it what you want to do?

Regards,
Benoit

KristianDukov, 2018-09-28:
Hi Benoit,

I just created a new object and added the things inside it.

```
JsonObject& merged = jsonBuffer.createObject();
merged["reps"] = n;
merged["power"] = powerContract;
merged["sets"] = s;                              
merged["contraction"] = contract;
merged["account"] = "gaga";
merged["machine"] = machine;
merged["date"] = "1990-06-01";

merged.printTo(buff_merged);
arr.add(buff_merged);
arr.printTo(bufferArray);
```

Now I just have to find a way to clear the `bufferArray` and start all over with the `arr`. Will check the `.clear()` for the `bufferArray`.
But thank you for the assistance! Great job creating this and supporting it!

Regards,
Kris