bblanchon/ArduinoJson #818: Having A Problem With Reading an array of jsonc objects - plz help

GhassanYusuf, 2018-09-24:
Dear Friend

i have the following json string :

[{"ssid":"BBIoT - Guest","bssid":"00:0D:67:7F:E5:34","password":"bahrainiothub","zone":"Teams"},{"ssid":"TAKEONE","bssid":"30:B5:C2:63:56:62","password":"$a$h1983","zone":"Teams"},{"ssid":"BBIoT","bssid":"00:0D:67:7F:E5:33","password":"bbiot888","zone":"Teams"},{"ssid":"Brinc Staff","bssid":"FC:EC:DA:17:F2:47","password":"88888888","zone":"Staff"}]

i am trying to read them with a for loop but i always get an error

void connectKnown(String wifiJSONList) 
{
    DynamicJsonBuffer   jsonBuffer;
    JsonObject& root  = jsonBuffer.parseObject(wifiJSONList);

    // Trying To Connect To Known WiFi
    for(int i=0; i<root.size(); i++) {
      if(connectWiFi(root[i]["ssid"], root[i]["password"])){
        break;
      }
    }
    delay(1000);
}

i just cant get it to work.

bblanchon, 2018-09-25:
Hi @GhassanYusuf,

From what I can see, you need to call [JsonBuffer::parseArray()](https://arduinojson.org/v5/api/jsonbuffer/parsearray/?utm_source=github&utm_medium=issues) instead of [JsonBuffer::parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/?utm_source=github&utm_medium=issues).

See also:

* [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/?utm_source=github&utm_medium=issues)
* [FAQ: Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)

Regards,
Benoit

GhassanYusuf, 2018-09-25:
Can you please show me how its done ?!

GhassanYusuf, 2018-09-25:
Could you show me an example ?!

bblanchon, 2018-09-26:
Hi @GhassanYusuf,

Here is the example from [JsonArray::begin() / JsonArray::end()](https://arduinojson.org/v5/api/jsonarray/begin_end/?utm_source=github&utm_medium=issues):

```c++
char json[] = "[\"one\",\"two\",\"three\"]";
DynamicJsonBuffer jsonBuffer;
JsonArray& arr = jsonBuffer.parseArray(json);

for (auto value : arr) {
    Serial.println(value.as<char*>());
}
```

Here is the program generated by the [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/?utm_source=github&utm_medium=issues):

```c++
const size_t bufferSize = JSON_ARRAY_SIZE(4) + 4*JSON_OBJECT_SIZE(4) + 310;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "[{\"ssid\":\"BBIoT - Guest\",\"bssid\":\"00:0D:67:7F:E5:34\",\"password\":\"bahrainiothub\",\"zone\":\"Teams\"},{\"ssid\":\"TAKEONE\",\"bssid\":\"30:B5:C2:63:56:62\",\"password\":\"$a$h1983\",\"zone\":\"Teams\"},{\"ssid\":\"BBIoT\",\"bssid\":\"00:0D:67:7F:E5:33\",\"password\":\"bbiot888\",\"zone\":\"Teams\"},{\"ssid\":\"Brinc Staff\",\"bssid\":\"FC:EC:DA:17:F2:47\",\"password\":\"88888888\",\"zone\":\"Staff\"}]";

JsonArray& root = jsonBuffer.parseArray(json);

JsonArray& root_ = root;

JsonObject& root_0 = root_[0];
const char* root_0_ssid = root_0["ssid"]; // "BBIoT - Guest"
const char* root_0_bssid = root_0["bssid"]; // "00:0D:67:7F:E5:34"
const char* root_0_password = root_0["password"]; // "bahrainiothub"
const char* root_0_zone = root_0["zone"]; // "Teams"

JsonObject& root_1 = root_[1];
const char* root_1_ssid = root_1["ssid"]; // "TAKEONE"
const char* root_1_bssid = root_1["bssid"]; // "30:B5:C2:63:56:62"
const char* root_1_password = root_1["password"]; // "$a$h1983"
const char* root_1_zone = root_1["zone"]; // "Teams"

JsonObject& root_2 = root_[2];
const char* root_2_ssid = root_2["ssid"]; // "BBIoT"
const char* root_2_bssid = root_2["bssid"]; // "00:0D:67:7F:E5:33"
const char* root_2_password = root_2["password"]; // "bbiot888"
const char* root_2_zone = root_2["zone"]; // "Teams"

JsonObject& root_3 = root_[3];
const char* root_3_ssid = root_3["ssid"]; // "Brinc Staff"
const char* root_3_bssid = root_3["bssid"]; // "FC:EC:DA:17:F2:47"
const char* root_3_password = root_3["password"]; // "88888888"
const char* root_3_zone = root_3["zone"]; // "Staff"
```

Combining them is left as an exercise for the reader :wink:

Regards,
Benoit

GhassanYusuf, 2018-09-26:
Thanks Man - Really Really Appreciate it 👍 😄