bblanchon/ArduinoJson #822: How use root globally?

dEROZA, 2018-09-29:
```
const size_t bufferSize = JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(8);
DynamicJsonBuffer jsonBuffer(bufferSize);

JsonObject& root = jsonBuffer.createObject();

JsonObject& conf = root.createNestedObject("conf");
JsonObject& ap = root.createNestedObject("ap");
JsonObject& sta = root.createNestedObject("sta");
```

and

`root = jsonBuffer.parseObject(f);`

> sketch_sep14a:137:8: error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
>    root = jsonBuffer.parseObject(f);
>         ^

dEROZA, 2018-10-01:
At first I builded object and convert it into a string..
```
String json;
root.prettyPrintTo(json);
return json;
```

but in another time I can parse it for next changes..
`JsonObject &root_d = jsonBuffer.parseObject(ConfigFile);`

closed 2018-10-01 by dEROZA

bblanchon, 2018-10-01:
Hi @dEROZA,

I'm sorry I didn't answer earlier, I somehow missed your question.

I recommend against using global `JsonBuffer` and `JsonObject` because the library is not optimized for that.
Version 6 allows to do that, but you should think twice before doing so because the `JsonObject` would eat a considerable amount of RAM.

If you need to keep the JSON document itself, serializing it to a string is a good idea, because it's more compact than the `JsonObject`.

However, if you're just interested in the values stored in the JSON document, I recommend that you create a custom `struct`, as shown in [JsonConfigFile.ino](https://arduinojson.org/v5/example/config/?utm_source=github&utm_medium=issues)

See also:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)

Regards,
Benoit

dEROZA, 2018-10-01:
SD conflict with FS, I use FS.. so many changes I write as you in JsonConfigFile.ino, so this is a general point in theme why I asked for a global jsonobj. But fix my problems, so thx a lot for your answer, because I fully understand how it works..