bblanchon/ArduinoJson #823: Parsing stopped working after some time

ukpiglet, 2018-09-30:
Arduino 1.8.7
Board: Generic ESP8266
ArduinoJson: 5.13.2
LoLin Nodemcu

Code:

```

#include <ArduinoJson.h>
const size_t bufferSize = JSON_OBJECT_SIZE(4) + 30;
DynamicJsonBuffer jsonBuffer(bufferSize);

(Data in "s" was sent to the ESP8266 by MQTT)

  Serial.println(" I got: ");
  Serial.println(s);
  
  JsonObject& root = jsonBuffer.parseObject(s);

  int r = root["r"];
  int g = root["g"];
  int b = root["b"];
  int n = root["n"];
  
  Serial.printf("Colour... r=%d g=%d b=%d n=%d\n", r, g, b, n);
```

To start off with parsing is fine:

> I got: 
> {"r":0,"g":10,"b":0,"n":8}
> Colour... r=0 g=10 b=0 n=8

After some time it parses all input as zero:

> I got: 
> {"r":0,"g":10,"b":0,"n":7}
> Colour... r=0 g=0 b=0 n=0

The parsing resulting in all zeroes continues until the board is reset, at which time it works OK for a while.

Any suggestions would be appreciated.

bblanchon, 2018-10-01:
Hi @ukpiglet,

The parsing fails because the [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues) is full.
You need to create a new one: just move the definition of the [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues) inside the function, and it should work.

See:

* [The first parsing succeeds, why do the next ones fail?](https://arduinojson.org/v5/faq/the-first-parsing-succeeds-why-do-the-next-ones-fail/?utm_source=github&utm_medium=issues)
* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)
* [Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)

Regards,
Benoit

ukpiglet, 2018-10-01:
Thank you. Yes! Makes perfect sense. I'll try it out tonight.

Not being particularly proficient in C I expected parseObject to reuse the buffer.