bblanchon/ArduinoJson #824: add jsonObject (from Class.function.return) to jsonArray failed

Pfannex, 2018-10-01:
Hi,

I got a Widget-class that holds some propertys. Also this Class got a function that returns a JsonObject& with these propertys.

```c++
  class Widget {
  public:
    Widget(String name);
    String name = "";
    String type = "";
    String caption = "";
    int readonly = 1;
    String inputtype = "";
    String direction = "";
    int min = 1;
    int max = 1;
    String event = "";
    String action = "";

    JsonObject& asObject();
  };
```
```c++
JsonObject& Widget::asObject(){
  const size_t bufferSize = JSON_OBJECT_SIZE(10);
  DynamicJsonBuffer jsonBuffer(bufferSize);
  JsonObject& root = jsonBuffer.createObject();
    root["name"]      = name;
    root["type"]      = type;
    root["caption"]   = caption;
    root["readonly"]  = readonly;
    root["inputtype"] = inputtype;
    root["direction"] = direction;
    root["min"]       = min;
    root["max"]       = max;

    root["event"]     = event;
    root["action"]    = action;
  return root;
}
```

The Dashboard-class is used to plug some widgets together to a JsonArray of JsonObjects. Finaly I need the assembled Json-String to write them into a file in the SPIFFS.
```c++
 String Dashboard::page_main(){  
   Widget w1("Widget_1");
    w1.type      = "text";
    w1.caption   = "NEWTime";
    w1.readonly  = 1;
    w1.event     = "~/event/clock/time";
    w1.inputtype = "datetime";

  w1.asObject().prettyPrintTo(Serial);
  Serial.println("-------------------------");

  Widget w2("Widget_2");
    w2.type      = "text";
    w2.caption   = "NEWNEWTime";
    w2.readonly  = 1;
    w2.event     = "~/event/clock/time";
    w2.inputtype = "datetime";

  w2.asObject().prettyPrintTo(Serial);
  Serial.println("-------------------------");

  DynamicJsonBuffer jsonBuffer;
  JsonArray& root = jsonBuffer.createArray();
  root.add(w1.asObject());
  root.add(w2.asObject());

  String rootStr;
  root.prettyPrintTo(Serial);
  Serial.println("-------------------------");
  root.printTo(rootStr);
  return rootStr;
```

`w1.asObject().prettyPrintTo(Serial);` returns correctly:
```
{
  "name": "Widget_1",    <----
  "type": "text",
  "caption": "NEWTime",   
  "readonly": 1,
  "inputtype": "datetime",

  "direction": "",
  "min": 1,
  "max": 1,
  "event": "~/event/clock/time",
  "action": ""
}
```
also `w1.asObject().prettyPrintTo(Serial);`
```
{
  "name": "Widget_2",    <----
  "type": "text",
  "caption": "NEWNEWTime", 
  "readonly": 1,
  "inputtype": "datetime",
  "direction": "",
  "min": 1,
  "max": 1,
  "event": "~/event/clock/time",
  "action": ""
}
```

**but**

`root.prettyPrintTo(Serial);` returns

```
[
  {
    "name": "Widget_2",    <----
    "type": "text",
    "caption": "NEWNEWTime",
    "readonly": 1,

    "inputtype": "datetime",
    "direction": "",
    "min": 1,
    "max": 1,
    "event": "~/event/clock/time",
    "action": ""
  },
  {
    "name": "Widget_2",    <----
    "type": "text",
    "caption": "NEWNEWTime",
    "readonly": 1,
    "inputtype": "datetime",
    "direction": "",
    "min": 1,
    "max": 1,
    "event": "~/event/clock/time",
    "action": ""
  }
]
```

and finaly

```c++
root.printTo(rootStr);
```
the sketch crashes.....

** So why is the widget inserted twice in the root array while the individual objects w1 and w2 seems to be correct? **

Thanks and Greatings from Hamburg
Pf@nne

bblanchon, 2018-10-02:
Hi @Pfannex,

The problem is that the function `Widget::asObject()` returns a reference to a destructed object.
Remember that the [`JsonObject`](https://arduinojson.org/v5/api/jsonobject/?utm_source=github&utm_medium=issues) is inside the [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues); so, when you destroy the [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues), you also destroy the [`JsonObject`](https://arduinojson.org/v5/api/jsonobject/?utm_source=github&utm_medium=issues).

Here are several ideas to solve this problem:

1. Return a string: `String asObject()`
2. Pass the [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues): `asObject(DynamicJsonBuffer&)`
3. Pass a [`JsonObject`](https://arduinojson.org/v5/api/jsonobject/?utm_source=github&utm_medium=issues): `asObject(JsonObject&)`

My favorite is the last one because it allows composing several independent objects in the same JSON document. It is the technique I used in the case studies of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues), for the JSON configuration file on SPIFFS.

Regards,
Benoit

Pfannex, 2018-10-02:
Hi Benoit,

again special thanks for your great quick support!
Meanwhile I spend my time to had a look into your Mastering ArduinoJson. Great useful stuff!

I changed my code to idea 3....
To pass a reference Object works as aspected.

```c++
void Widget::asObject(JsonObject& root){
  root["name"]      = name;
  root["type"]      = type;
  root["caption"]   = caption;
  root["readonly"]  = readonly;
  root["inputtype"] = inputtype;
  root["direction"] = direction;
  root["min"]       = min;
  root["max"]       = max;

  root["event"]     = event;
  root["action"]    = action;
}
```
```c++
String Dashboard::page_main(){
//page_main root json
  DynamicJsonBuffer root_jsonBuffer;
  JsonArray& root = root_jsonBuffer.createArray();
//widget size
  const size_t bufferSize = JSON_OBJECT_SIZE(10);

//widget 1
  DynamicJsonBuffer w1_jsonBuffer(bufferSize);
  JsonObject& w1_root = w1_jsonBuffer.createObject();
  Widget w1("Widget_1");
    w1.type      = "text";
    w1.caption   = "Time 1";
    w1.readonly  = 1;
    w1.event     = "~/event/clock/time";
    w1.inputtype = "datetime";
  w1.asObject(w1_root);
  root.add(w1_root);

//widget 2
  DynamicJsonBuffer w2_jsonBuffer(bufferSize);
  JsonObject& w2_root = w2_jsonBuffer.createObject();
  Widget w2("Widget_2");
    w2.type      = "text";
    w2.caption   = "Time 2";
    w2.readonly  = 1;
    w2.event     = "~/event/clock/time";
    w2.inputtype = "datetime";
  w2.asObject(w2_root);
  root.add(w2_root);

// return main_page rootArray as String
  String rootStr;
  root.printTo(rootStr);
  return rootStr;
}
```
So I need to declare a jsonBuffer for each new object, because the .add to the rootArray is again a reference to the jsonBuffer. 
That causes, that the widget class is quite empty....
So it can be contained into the dashboard class.

Therefore I consider to switch to the string variation.
Because copies are possible....
Main focus should be the assembly of the rootArray (.json File).

For explanation:
For our [ESP8266 OmniESP-framework](https://github.com/Pfannex/OmniESP/wiki) we use your ArduinoJson to handle our SPIFFS JsonFiles. 
One of them holds our WebIF-Dashboard for showing measurement data or controlling some hardware switches.
The dashboardJson is a Array of Widget-Objects witch can be restructured at runtime.
Therefor we need a efficent and good looking class, thas what I´m testing right now....

Maybe you got a inspiration..

Greetings
Marco

Pfannex, 2018-10-02:
Hi Benoit,

meanwhile I got a look into your Mastering ArduinoJson, great stuff!
I change the code to variante 3.
Everything works as expected!

```c++
void Widget::asObject(JsonObject& root){
  root["name"]      = name;
  root["type"]      = type;
  root["caption"]   = caption;
  root["readonly"]  = readonly;
  root["inputtype"] = inputtype;
  root["direction"] = direction;
  root["min"]       = min;
  root["max"]       = max;

  root["event"]     = event;
  root["action"]    = action;
}
```
```c++
String Dashboard::page_main(){
//page_main root json
  DynamicJsonBuffer root_jsonBuffer;
  JsonArray& root = root_jsonBuffer.createArray();
//widget size
  const size_t bufferSize = JSON_OBJECT_SIZE(10);

//widget 1
  DynamicJsonBuffer w1_jsonBuffer(bufferSize);
  JsonObject& w1_root = w1_jsonBuffer.createObject();
  Widget w1("Widget_1");
    w1.type      = "text";
    w1.caption   = "Time 1";
    w1.readonly  = 1;
    w1.event     = "~/event/clock/time";
    w1.inputtype = "datetime";
  w1.asObject(w1_root);
  root.add(w1_root);

//widget 2
  DynamicJsonBuffer w2_jsonBuffer(bufferSize);
  JsonObject& w2_root = w2_jsonBuffer.createObject();
  Widget w2("Widget_2");
    w2.type      = "text";
    w2.caption   = "Time 2";
    w2.readonly  = 1;
    w2.event     = "~/event/clock/time";
    w2.inputtype = "datetime";
  w2.asObject(w2_root);
  root.add(w2_root);

// return main_page rootArray as String
  String rootStr;
  root.printTo(rootStr);
  return rootStr;
}
```
That makes the widget class a little bit empty....so the rest can also be contained into the dashboard class.

The main focus should be the easy and good looking assembly of the rootArray, that represents the dashboard.json.

For explanation:
For our [ESP8266 OmniESP-framework](https://github.com/Pfannex/OmniESP/wiki) we use your ArduinoJson to handle our SPIFFS JsonFiles. 
One of them holds our WebIF-Dashboard for showing measurement data or controlling some hardware switches.
The dashboardJson is a Array of Widget-Objects witch can be restructured at runtime.
Therefor we need a efficent and good looking class, thats what I´m testing right now....

Maybe you got a inspiration..

Greetings
Marco

bblanchon, 2018-10-03:
Hi Marco,

This design makes sense; I don't think that the fact that `Widget` has few responsibilities is a bad thing.
On the contrary, it provides a strong-typed wrapper on top of the sloppy JSON format.

Also, if the platform evolves, you could make `Widget` polymorphic, so that different classes have different configurations.

Concerning the usage of ArduinoJson, you just need one `DynamicJsonBuffer`:

```c++
String Dashboard::page_main(){
  size_t bufferSize = 2 * JSON_OBJECT_SIZE(10);
  DynamicJsonBuffer jsonBuffer(bufferSize);
  JsonArray& root = jsonBuffer.createArray();

  JsonObject& w1_root = root.createNestedObject();
  Widget w1("Widget_1");
    w1.type      = "text";
    w1.caption   = "Time 1";
    w1.readonly  = 1;
    w1.event     = "~/event/clock/time";
    w1.inputtype = "datetime";
  w1.asObject(w1_root);

  JsonObject& w2_root = root.createNestedObject();
  Widget w2("Widget_2");
    w2.type      = "text";
    w2.caption   = "Time 2";
    w2.readonly  = 1;
    w2.event     = "~/event/clock/time";
    w2.inputtype = "datetime";
  w2.asObject(w2_root);

  String rootStr;
  root.printTo(rootStr);
  return rootStr;
}
```

Regards,
Benoit

Pfannex, 2018-10-03:
Moin Benoit,

sometime I don't see the forest for the trees!
To declare the widget-object as nestedObject of the rootArray is pretty smarter and more readable. 
I think that we found, again with your help,  the final structure of our dashboard.json assembler!

Thanks and good job...
Marco

Pfannex, 2018-10-03:
The problem is how to handle the objects within the iteration.
I guess it isn't possible to create a jsonObject inside a loop?
But it seems to work fine.

What does `JsonArray& gridDataRow = gridData.createNestedArray();` do in the loop? 

```c++
//create gridObjects
  JsonObject& grid = root.createNestedObject();
    grid["type"] = "grid";
    grid["name"] = "grid1";

//create grid
  //fill GridData
  JsonArray& gridData = grid.createNestedArray("data");
  int i = 0;
  for (auto &element : rootCFG) {
    JsonArray& gridDataRow = gridData.createNestedArray();
    //column 1 "key"
    JsonObject& keyObject = gridDataRow.createNestedObject();
      keyObject["type"]     = "text";
      keyObject["name"]     = "c1" + String(i);
      keyObject["caption"]  = element.key;
      keyObject["readonly"] =  1;
      keyObject["event"]    =  "";
      keyObject["prefix"]   =  "";
      keyObject["suffix"]   =  "";
    i++;
  }
```

Greetings
Marco

bblanchon, 2018-10-04:
Hi Marco,

There is nothing wrong with calling `createNestedArray()` in a loop.
Your program looks correct, am I missing something?

Regards,
Benoit

Pfannex, 2018-10-10:
Hi Benoit,

I was a little bit irritated to create objects with the same Name in a loop.

Now I hope that I understand why it works.
`gridDataRow` and `keyObject` are references (aliases) to a memory address in the jsonBuffer. With .createObject some more memory is allocatet and the reference to this address ist stored in the reference.
In the next loop the reference points to the next Object.
After leaving the loop gridDataRow is the reference to the last created object.

Regards 
Marco

bblanchon, 2018-10-11:
Hi Marco,

It seems that you understand correctly.

However, I'm puzzled by your last sentence: after leaving the loop, the reference `gridDataRow` doesn't exist anymore. The `JsonObject` is still in the `JsonBuffer`, but the reference is gone, you cannot use it out of the block. In other words, the scope of `gridDataRow` is the block.

Regards,
Benoit

closed 2018-10-24 by bblanchon