bblanchon/ArduinoJson #825: Formatting structs for serializing JSON

KentWalker, 2018-10-04:
Platform: Arduino Mega with Arduino Ethernet 2 with SD card.

I have a project where settings will be stored on a SD card but may be edited from an HTML page. The Mega hosts a webserver that provides the setup webpages.

I have JavaScript that retrieves data from forms and produces JSON objects which are saved directly as json files on the SD card.

Looking at your SpiffsConfig example, I would like to do something similar to allow for save/load functions in the Mega.  I can follow how the example works but I am not sure how to create structures that will interface to the JavaScript generated files. Please note that the JavaScript treats all form settings as character strings (it uses JSON.stringify() to package the forms).  Since the forms contain a variety of checkboxes, radio buttons, and text/number fields, sometimes the json files created aren't formatted as efficiently as they could but the Javascript works across all forms ok. 

For example, the JavaScript created "bridge.jsn" file contains:
`
{
	"trigger":"2",
	"bridge1":[ "1", "4", "15000", "50", "800", "0" ],
	"bridge2":[ "2", "4", "0", "50", "0", "1" ],
	"bridge3":[ "0", "1", "0", "0", "0", "0" ],
	"bridge4":[ "0", "1", "0", "0", "0", "0" ],
	"option":[ "250", "30", "40", "0", "1", "3", "2", "0.5" ]
}`

I have defined my bridge structure in the header file:
`struct BrConfig {
  char port[1];
  char baud[1];
  char trip[5];
  char min[5];
  char test[5];
  char zero[1];

  void load(const JsonObject &);
  void save(JsonObject &) const;
};

struct OpConfig {
  char under[3];
  char utmo[3];
  char zb[3];
  char negative[1];
  char reset[1];
  char aReset[4];
  char zero[1];
  char aZero[4];

  void load(const JsonObject &);
  void save(JsonObject &) const;
};

struct BridgeConfig {
  char trigger[1];
  BrConfig bridge1;
  BrConfig bridge2;
  BrConfig bridge3;
  BrConfig bridge4;
  OpConfig options;

  void load(const JsonObject &);
  void save(JsonObject &) const;
}

bool serializeBridgeConfig(const BridgeConfig &config, Print &dst);
bool deserializeBridgeConfig(Stream &src, BridgeConfig &config);
`

When I come to save and load, I have something like this in the cpp file:  
`const size_t bufferSize = 4*JSON_ARRAY_SIZE(6) + JSON_ARRAY_SIZE(8) + JSON_OBJECT_SIZE(6);

void OpConfig::save(JsonObject &obj) const {
  obj[0] = under;
  obj[1] = utmo;
  obj[2] = zb;
  obj[3] = negative;
  obj[4] = reset;
  obj[5] = aReset;
  obj[6] = zero;
  obj[7] = aZero;
}

void OpConfig::load(const JsonObject &obj) {
  strlcpy(under, obj[0]| "0", sizeof(under));
  strlcpy(utmo, obj[1]| "0", sizeof(utmo));
  strlcpy(zb, ob[2]| "0", sizeof(zb));
  strlcpy(negative, obj[3]| "0", sizeof(negative));
  strlcpy(reset, obj[4]| "0", sizeof(reset));
  strlcpy(aReset, obj[5]| "0", sizeof(aReset));
  strlcpy(under, obj[6]| "0", sizeof(zero));
  strlcpy(under, obj[7]| "0", sizeof(aZero));
}

void BrConfig::save(JsonObject &obj) const {
  obj[0] = port;
  obj[1] = baud;
  obj[2] = trip;
  obj[3] = min;
  obj[4] = test;
  obj[5] = zero;
}

void BrConfig::load(const JsonObject &obj) {
  strlcpy(port, obj[0]| "0", sizeof(port));
  strlcpy(baud, obj[1]| "0", sizeof(baud));
  strlcpy(trip, ob[2]| "0", sizeof(trip));
  strlcpy(min, obj[3]| "0", sizeof(min));
  strlcpy(test, obj[4]| "0", sizeof(test));
  strlcpy(zero, obj[5]| "0", sizeof(zero));
}

void BridgeConfig::save(JsonObject &obj) const {
  obj["trigger"] = trigger;
  bridge1.save(obj.createNestedObject("bridge1"));
  bridge2.save(obj.createNestedObject("bridge2"));
  bridge3.save(obj.createNestedObject("bridge3"));
  bridge4.save(obj.createNestedObject("bridge4"));
  options.save(obj.createNestedObject("option"));
}

void BridgeConfig::load(const JsonObject &obj) {
  strlcpy(trigger, obj["trigger"]| "0", sizeof(trigger));
  bridge1.load(obj["bridge1"]);
  bridge2.load(obj["bridge2"]);
  bridge3.load(obj["bridge3"]);
  bridge4.load(obj["bridge4"]);
  options.load(obj["option"]);
}

bool serializeBridgeConfig(const BridgeConfig &config, Print &dst) {
  StaticJsonBuffer jb(bufferSize);
  JsonObject &root = jb.createObject();
  config.save(root);
  return root.printTo(dst);
}

bool deserializeBridgeConfig(Stream &src, BridgeConfig &config) {
  StaticJsonBuffer jb(bufferSize + 150);
  JsonObject &root = jb.parseObject(src);
  if (!root.success()) return false;
  config.load(root);
  return true;
}
`

Finally in the main part of the code, I only need the load file section, since the controller isn't responsible for file saving or creation.

`const char *bridgeConfigFile = "bridge.jsn";

void getBridgeConfig() {
  BridgeConfig bridgeConfig;
  File file = SD.open(bridgeConfigFile, FILE_READ);
  if(deserializeBridgeConfig(file, config)) {
    // do stuff with the loaded config 
  }
}
`

Would this the the most efficient way of doing this?  As the project resides on an Arduino Mega, I'm concerned about hitting RAM limits.

bblanchon, 2018-10-04:
Hi @KentWalker,

The JSON document is quite small so RAM should not be an issue.
However, the code presented above uses `JsonObject` where the JSON document has arrays.

For example:

```c++
void OpConfig::save(JsonObject &obj) const {
obj[0] = under;
obj[1] = utmo;
obj[2] = zb;
obj[3] = negative;
obj[4] = reset;
obj[5] = aReset;
obj[6] = zero;
obj[7] = aZero;
}
```

Should be

```c++
void OpConfig::save(JsonArray &arr) const {
arr.add(under);
arr.add(utmo);
arr.add(zb);
arr.add(negative);
arr.add(reset);
arr.add(aReset);
arr.add(zero);
arr.add(aZero);
}
```

Finally, I find it silly to use strings instead of integer and float.
Not only the JSON document is bigger, but you also lose all the value types and the code on the Arduino becomes overly complicated.
I understand that it simplifies the JavaScript side, but is it really where you want the code to be simple?
If I were you, I would do as much preprocessing on the client side so as to keep the Arduino code as simple as possible.

Regards,
Benoit

KentWalker, 2018-10-05:
Hi @bblanchon,

Thank you so much.  That makes much more sense using a JsonArray to save the structures.

So now I have the following code to save to a JSON file:
`void BrConfig::save(JsonArray &arr) const {
  arr.add(port);
  arr.add(baud);
  arr.add(trip);
  arr.add(minWt);
  arr.add(test);
  arr.add(zero);
}

void OpConfig::save(JsonArray &arr) const {
  arr.add(under);
  arr.add(utmo);
  arr.add(zb);
  arr.add(negative);
  arr.add(rReset);
  arr.add(aReset);
  arr.add(rZero);
  arr.add(aZero);
}

void BridgeConfig::save(JsonObject &obj) const {
  obj["trigger"] = trigger;
  bridge1.save(obj.createNestedArray("bridge1"));
  bridge2.save(obj.createNestedArray("bridge2"));
  bridge3.save(obj.createNestedArray("bridge3"));
  bridge4.save(obj.createNestedArray("bridge4"));
  options.save(obj.createNestedArray("option"));
}

etc`

To load the struc would I still use objects? I think this is correct: 
`void BrConfig::load(const JsonObject &obj) {
  obj[0] = port;
  obj[1] = baud;
  obj[2] = trip;
  obj[3] = minWt;
  obj[4] = test;
  obj[5] = zero;
}

void OpConfig::load(const JsonObject &obj) {
  obj[0] = under;
  obj[1] = utmo;
  obj[2] = zb;
  obj[3] = negative;
  obj[4] = rReset;
  obj[5] = aReset;
  obj[6] = rZero;
  obj[7] = aZero;
}

void BridgeConfig::load(const JsonObject &obj) {
  obj["trigger"] = trigger;
  bridge1.load(obj["bridge1"]);
  bridge2.load(obj["bridge2"]);
  bridge3.load(obj["bridge3"]);
  bridge4.load(obj["bridge4"]);
  options.load(obj["option"]);
}`

or is there a corresponding way of extracting a JsonArray that I am missing?

I was using strings only for proof of concept (JavaScript tends to default form data as strings), and it also makes it easier to debug.  Printing a null string shows "", an int with a null value shows as 0 (but it might actually, really be 0) but try indexing an array with a null...  
However I have spent the afternoon bashing the JavaScript into producing nice and safe JSON files.  

The bridge.jsn file now contains:
`{
	"trigger":2,
	"bridge1":[ 1, 4, 15000, 50, 800, 0 ],
	"bridge2":[ 2, 4, 0, 50, 0, 1 ],
	"bridge3":[ 0, 1, 0, 0, 0, 0 ],
	"bridge4":[ 0, 1, 0, 0, 0, 0 ],
	"option":[ 250, 30, 40, 0, 1, 3, 2, 0.5 ]
}`
and the structs now cast into the appropriate data types.

bblanchon, 2018-10-06:
Hi @KentWalker,

You have to use `JsonArray` for the load too, because `obj[0]` will not compile.

Regards,
Benoit

closed 2018-10-24 by bblanchon

schmonet, 2019-01-16:
Hi Benoit,

1
2
`12`

i have a similar question, so will add it to this thread.
I need to serialize and deserialize some structs conataining various elements, which should be added by the user in the Web-UI and sent back to the ESP8266 to be stored in the persistantly eeprom as struct or on the SPIFFS as JSON. I bought your eBook. Good work!

I have these struc and array to provide a preset of color and control-settings for my lamp, which is supposed to behave (soft on, soft of, motion detector) differently, depending on the time of day (NTP):

`typedef struct {
	uint32_t time; //time to next dimming point
	uint16_t h, s, v, w, ww; //lamp color, sat, hue and extra white + warm white
} dim_step_t;

/*
dim_table 'time table' to specify dim_table
*/
typedef struct
{
  uint32_t clock; //time of day
  uint16_t stdby, dist, ldr; //standby time, motion detection distance, ambient light level threshold
  dim_step_t dim_step[4]; //as a preset only 4 dimming points will be populated, but the user should be able to add more or delete some
} dim_table_t;

dim_table_t dim_table[] = {
   //clock, stdby, dist, ldr, time, h, s, v, w, ww, time, h, s, v, w, ww, time, h, s, v, w, ww...
	{ 480, 20,  0,  50, {{0, 0, 0, 0, 0, 0},{3, 0, 0, 500, 0, 0},{30, 0, 0,  50, 0, 0},{ 30, 0, 0, 0, 0, 0}}},
	{ 720,  0,  1, 100, {{0, 0, 0, 0, 0, 0},{1, 0, 0, 100, 0, 0},{20, 0, 0, 100, 0, 0},{ 10, 0, 0, 0, 0, 0}}},
	{1080,  0,  1,  50, {{0, 0, 0, 0, 0, 0},{3, 0, 0,  30, 0, 0},{10, 0, 0,  30, 0, 0},{300, 0, 0, 0, 0, 0}}},
	{1260,  1, 20,   0, {{0, 0, 0, 0, 0, 0},{6, 0, 0,  10, 0, 0},{60, 0, 0,  10, 0, 0},{ 10, 0, 0, 0, 0, 0}}},
}; //clock = 480 defines the first timeslot in minutes of day from midnight in relation to NTP wall clock time`

this should be serialized with ArduinoJson to:

`[
  { 
    "clock": 480, 
    "stdby": 20,
    "dist": 0, 
    "ldr": 50,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 500, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 50, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 30, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 720, 
    "stdby": 0,
    "dist": 1, 
    "ldr": 100,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 1, 
        "h": 0, 
        "s": 0, 
        "v": 100, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 20, 
        "h": 0, 
        "s": 0, 
        "v": 100, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 1080, 
    "stdby": 0,
    "dist": 1, 
    "ldr": 50,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 30, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 30, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 300, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 1260, 
    "stdby": 1,
    "dist": 20, 
    "ldr": 0,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 6, 
        "h": 0, 
        "s": 0, 
        "v": 10, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 60, 
        "h": 0, 
        "s": 0, 
        "v": 10, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  }
]`

I used https://arduinojson.org/v5/assistant/ to calculate the Buffersize for ESP8266 to 2916:

`const size_t bufferSize = 5*JSON_ARRAY_SIZE(4) + 4*JSON_OBJECT_SIZE(5) + 16*JSON_OBJECT_SIZE(6) + 620;`

i wanted to serialize dim_table[] as follows:

`DynamicJsonBuffer jsonBuffer(bufferSize);
JsonArray& root = jsonBuffer.createArray();
String json; 

for (uint8_t i = 0; i < (4 - 1); i++) { 
	JsonObject& block = root.createNestedObject("block");
	block["clock"] = dim_table[i].clock;
	block["stdby"] = dim_table[i].stdby;
	block["dist"] = dim_table[i].dist;
	block["ldr"] = dim_table[i].ldr;	
	for (uint8_t j = 0; j < (4 - 1); j++) {				
		JsonObject& step = line.createNestedObject("step");
		step["time"] = dim_table[i].dim_step[j].time;
		step["h"] = dim_table[i].dim_step[j].h;
		step["s"] = dim_table[i].dim_step[j].s;
		step["v"] = dim_table[i].dim_step[j].v;
		step["w"] = dim_table[i].dim_step[j].w;
		step["ww"] = dim_table[i].dim_step[j].ww;
	}
} //i,j < (4 - 1) needs to be changed to dynamically adapt to more ore less nodes, if the user will delete some or add some in the WebUI`

but i assume, i am supposed to use "add" ?! I am also wondering, if i better should go for the new jsonDocument with v6 of arduinoJson?

Any comments are welcome. thanks!

bblanchon, 2019-01-17:
Hello @schmonet,

Thank you very much for purchasing the eBook 👍 

Please, open a new issue on GitHub so I can keep track of it.

I've read your message several times but I don't understand what you're asking.
Can you state your question more clearly?

Thank you for your understanding
Benoit