bblanchon/ArduinoJson #826: Program size grows a lot with parsing

StarlightNL, 2018-10-04:
I am trying to parse a [small settings file](https://github.com/StarlightNL/KeyboardProject/blob/master/Misc/settings.json) that I calculated with the assistant. The parsing happens in [this](https://github.com/StarlightNL/KeyboardProject/blob/master/ComboProcessor/SettingsHelper.cpp) file.

When I use the parseObject function with a file the program size increases from 71.832 bytes to 129.688 bytes which is 99% of my MCU(Bluepill). The STM32 core I use is [this](https://github.com/rogerclarkmelbourne/Arduino_STM32) one. I'm not really a good question asker but hopefully it is clear enough.

bblanchon, 2018-10-06:
Hi @StarlightNL,

I worked very hard to keep the code as small as possible.
Since I only optimized for AVR code because it's the MCU with the less RAM, there may be something unexpected on STM32.

However, before jumping to conclusions, we should first look at the assembler.
I wrote a tutorial to see the size of each function:
https://blog.benoitblanchon.fr/arduino-code-size-and-assembly/
The article covers AVR, but you can probably adapt it for STM32.

Regards,
Benoit

StarlightNL, 2018-10-06:
Hi @bblanchon ,
I checked out the code size with the arm equivalent nm command.
Included are the 2 files with the parse function and without the parse function.
[logWithoutParse.txt](https://github.com/bblanchon/ArduinoJson/files/2453180/logWithoutParse.txt)
[logWithParse.txt](https://github.com/bblanchon/ArduinoJson/files/2453181/logWithParse.txt)

All the direct library functions are pretty low in the list but the other things it calls are not.
I can include the build files and the compiler output of both states if needed.

StarlightNL, 2018-10-08:
Hi @bblanchon ,

Fix one did it. The size has decreased A LOT. From  129.688 (99%) to only 81.016 (62%)

Fix two will not do it for my project because I need USB support.

Fix three failed with the error `FixedSizeFlashString.hpp: 18:45: error: 'strncmp_P' was not declared in this scope`
But that is a problem in the stm32 core I use. 

Since fix one is just in the platform.txt it will not break a lot of things so it is the best solution.

bblanchon, 2018-10-09:
Hi @StarlightNL,

Cool. It may be a good idea to make a pull request to add this flag upstream.

BTW, to fix `error: 'strncmp_P' was not declared in this scope`, you could disable support for Flash string by defining [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/?utm_source=github&utm_medium=issues) to 0.

Regards,
Benoit

closed 2018-10-09 by bblanchon