bblanchon/ArduinoJson #827: How to correct return JsonObject from my custom class?

VadimShtukan, 2018-10-12:
I cannot correct return JsonObject from my custom class. When I call serializeJson it calls exception. Inside class function serializeJson works correctly. 

# Error: 
```
Exception (3):
epc1=0x40202d08 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402032a4 depc=0x00000000

ctx: cont 
sp: 3ffffd40 end: 3fffffd0 offset: 01a0

```

# My code:
## SMConfigESP.h
```C++
#ifndef ARDUINOPROJECTS_CONFIGESP_H
#define ARDUINOPROJECTS_CONFIGESP_H

#include <ArduinoJson.h>

#define CONFIG_JSON_SIZE 200 //Размер json config ----

class SMConfigESP{
public:
    JsonObject buildJson(); 
};

#endif //ARDUINOPROJECTS_CONFIGESP_H

```

## SMConfigESP.cpp
```C++
#include "SMConfigESP.h"

SMConfigESP::SMConfigESP() {

}

JsonObject SMConfigESP::buildJson() {
    StaticJsonDocument<CONFIG_JSON_SIZE> jsonDoc;
    JsonObject root = jsonDoc.to<JsonObject>();

    root["sensor"] = "gps";
    root["time"] = 1351824120;

    JsonArray data = root.createNestedArray("data");
    data.add(48.756080);
    data.add(2.302038);
        
    String t;
    //It`s working!!!
    serializeJson(root, t);

    return root;
}
```

## main.ino
```C++
#include <ArduinoJson.h>
#include "SMConfigESP.h"

SMConfigESP smConfigESP;

void setup() {
  JsonObject rootJsonConfig = smConfigESP.buildJson();
  
  String t;
  //It throw exception!
  serializeJson(rootJsonConfig, t);
}

void loop() { 
}
```

bblanchon, 2018-10-12:
Hi @VadimShtukan,

`JsonObject` is a reference that points to the data in the `JsonDocument`.

There are several ways to solve this problem:

1. return the `JsonDocument` instead of the `JsonObject`
2. pass the  `JsonDocument` by reference
3. pass the `JsonObject` to the function

I prefer solution 3 because it allows composing a large JSON document by calling several builders functions. I explain how to use this pattern in the case study "Configuration in SPIFFS" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues)

However, I'll show you how to implement solution 1 because I think I never described it before.
You're using an ESP so you could safely switch to a `DynamicJsonDocument`, but I'll assume that you want to keep the `StaticJsonDocument`.

Here you go:

## SMConfigESP.h

```C++
#include <ArduinoJson.h>

#define CONFIG_JSON_SIZE 200 //Размер json config ----

class SMConfigESP{
public:
    using JsonDocument = StaticJsonDocument<CONFIG_JSON_SIZE>;
    JsonDocument buildJson(); 
};
```

## SMConfigESP.cpp

```C++
#include "SMConfigESP.h"

SMConfigESP::JsonDocument SMConfigESP::buildJson() {
    JsonDocument jsonDoc;
    JsonObject root = jsonDoc.to<JsonObject>();

    root["sensor"] = "gps";
    root["time"] = 1351824120;

    JsonArray data = root.createNestedArray("data");
    data.add(48.756080);
    data.add(2.302038);
        
    return jsonDoc;
}
```

## main.ino

```C++
#include <ArduinoJson.h>
#include "SMConfigESP.h"

SMConfigESP smConfigESP;

void setup() {
  auto rootJsonConfig = smConfigESP.buildJson();
  
  String t;
  serializeJson(rootJsonConfig, t);
}

void loop() { 
}
```

bblanchon, 2018-10-12:
Sorry, I went to fast with this answer.
Copying `JsonDocument`s is not possible right now, so you need to set solution 3 aside.
I'll work on it ASAP.

bblanchon, 2018-10-13:
I just published the [version 6.5.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.5.0-beta) that allows copying `JsonDocument`s.

VadimShtukan, 2018-10-16:
Ok, thank`s, I will try.

closed 2018-10-24 by bblanchon