bblanchon/ArduinoJson #829: Using an api key with ArduinoJson? Why isn't "Authentication: Basic" working?

arcbtc, 2018-10-15:
I have been trying to connect to this API for a good few days 

https://strike.acinq.co/documentation/api-reference

The key I have works fine in curl requests through terminal, but I just cant get the request to work with ArduinoJson. It connects, but then return "401 Unauthorised", which the API docs claim is an API key rejection, however I am 100% sure the key works?

Any help or suggestions would be very gratefully received?

    String PostData = "amount=1000&currency=\"btc\"&description=\"sweetsandthat\"";
  
    client.println("POST /api/v1/charges HTTP/1.1"); 
    client.println("Content-Type: application/x-www-form-urlencoded");
    client.println("Host: api.strike.acinq.co");
    client.println("Connection: keep-alive"); 
    client.println("Authorization: Basic "+apiKey);
    client.print("Content-Length: ");
    client.println(PostData.length());
    client.println(); // blank line required
    client.println(PostData);

    delay(1000); 
    String line = ""; 
    while (client.connected()) { 
       line = client.readStringUntil('999'); 
       Serial.println(line); 
    }

arcbtc, 2018-10-15:
The curl request I am trying to simulate is this one. As I said my API key works fine in terminal. 

    $ curl https://api.dev.strike.acinq.co/api/v1/charges \
       -u apiKey: \
       -X POST \
       -d amount=1000 \
       -d currency="btc" \
       -d description="sweetsandthat"

bblanchon, 2018-10-15:
Hi @arcbtc,

Please realize that your question is not related to ArduinoJson, the JSON library for Arduino, so my answer will be inaccurate.

Are you using HTTPS?
Did you try to remove the quotes in `PostData`?
Did you try to run curl in verbose mode to check what is sent?

I hope this answer will help you, but I have to close this issue as your question is not related to ArduinoJson.
Thank you for your understanding.

Regards,
Benoit

closed 2018-10-15 by bblanchon