bblanchon/ArduinoJson #831: ESP 32 Crashing loop

adolfie01, 2018-10-18:
hi

thank you for a good library. 

i am just using a plain out of the box esp 32 dev board, and only have 1 function in it.  when i run it, it seems to print the Json string then crash, and then loops like that.  i have tried root.printTo(Serial) and the way i did it below with the same error.

i cant see what is wrong, and why i get this. 

void setup()
{
Serial2.begin( 9200 ); // start GPS Serial Port
Serial.begin(115200);
}

void loop()
{
 BuildJson();
 
}

String BuildJson ()
{

const size_t bufferSize = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(8);
DynamicJsonBuffer jsonBuffer(bufferSize);

JsonObject& root = jsonBuffer.createObject();

root["DeviceID"] = DeviceID;
root["Sensor"] = "GPS";
root["Time"] = 1351824120;
root["Date"] = "18/10/2018";
root["Lat"] = "Some Lat";
root["Long"] = "Somelong";
root["Sensor1"] = "Some sensor 1 value";
root["Sensor2"] = "Some Sensor 2 Value";

String Json = "";

root.printTo(Json);
Serial.println(Json);
}

the error message seems to say "corrupt heap: bad head at 0x3ffb1f98, Expected 0xabba1234 got 0x0000000

any ideas

bblanchon, 2018-10-19:
Hi @adolfie01,

I copy-pasted your program and compiled it for my ESP8266.
It worked like a charm.

[image]

To compile the code, I made two adjustments: 

1. I removed `Serial2`
2. I replaced `DeviceID` with an integer 

I suspect there could be a bug in the ESP32 core.
Please make sure it's up to date.

Regards,
Benoit

adolfie01, 2018-10-19:
Hi

Thanks for this, it seems that it was a core update, I just deleted the old cores and reinstalled the files, now it works.

Thank you so very much for your help.

Gary

[image]<https://user-images.githubusercontent.com/5462433/47202944-04d25280-d37f-11e8-8d90-defdab85e187.png>

bblanchon, 2018-10-19:
You're welcome Gary.
Thank you for using ArduinoJson.
If you like this project, please [cast a start ⭐️](https://github.com/bblanchon/ArduinoJson/stargazers).

closed 2018-10-19 by bblanchon

adolfie01, 2018-10-20:
Hi

We are going to have to reopen this thread.

If you do anything else, that uses memory in the program, then as soon as it creates the Json array then crashes.

Is there not a Way to clear this buffer ?

My desktop name is not Ficeto, using the olmex EVB esp 32.

Any ideas

[cid:image001.png@01D46860.12DE88B0]

Kind Regards

Gary Jurgens

bblanchon, 2018-10-21:
Hi @adolfie01,

`JsonBuffer` is a throwaway memory pool, you're not supposed to reuse it.
Simply create a new `JsonBuffer` at each iteration.

See also:

* [How to reuse a `JsonBuffer`?](https://arduinojson.org/v5/faq/how-to-reuse-a-jsonbuffer/?utm_source=github&utm_medium=issues)
* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)

Regards,
Benoit

adolfie01, 2018-10-28:
Hi

Just so you know I solved the error,

Basically I had a method/function called "createJson()" it returned a string so String createJson(), the minute I put other code in the program it started its heap error again.

So I changed createJson() function to a Void. So "void createJson()" now it works.

Then I created a global variable "String JsonString = " ";

In function "createJson" I write the result to this JsonString variable And it works. However it does seem to add multiple jsonResults together. So you get these long lines of repeating Jsons.

I solved this by resetting the global variable JsonString to =" ". As the first line in createJson () function.

Now it all works 100% as designed.

Can call createJson() as needed even on an event or timer.

It's a strange work around, but it now works on the ESP32.

Maybe it helps someone :)

Thank you again for your assistance, it is greatly appreciated.

Thanks Gary