bblanchon/ArduinoJson #833: Most recent versions not published in as Particle.io library

damonmaria, 2018-10-23:
Although `library.properties` (which is for particle.io I'm pretty sure) has been updated with the most recent version (currently that file has `version=5.13.3`) it doesn't appear to be published:

```shell
$ particle library search arduinojson
> Found 1 library matching arduinojson
ArduinoJson 5.11.2 95233 An efficient and elegant JSON library for Arduino.
```

I'd love to use the new features in 5.12. Could this please be published? I think all you have to do (once the [Particle CLI is installed](https://docs.particle.io/guide/tools-and-features/cli/electron/#installing)) is:
```shell
$ particle library upload && particle library publish ArduinoJson
```

Full instructions: https://docs.particle.io/guide/tools-and-features/libraries/#publish-to-the-public

bblanchon, 2018-10-30:
Done!

Thank you very much @damonmaria for giving the instructions to publish the library: you saved me a lot of time!

closed 2018-10-30 by bblanchon

damonmaria, 2018-10-31:
Can confirm updated library works. Thanks!