bblanchon/ArduinoJson #834: reusing jsonBuffer in ESP8266

mTny, 2018-10-24:
Hello, i try to reuse same buffer every time i use a function, but for some reason after 4 times of calling the function the buffer became NULL,
i declate DynamicJsonBuffer inside ConnectAndCalculate funcation and in void loop i call it.
here is my code:

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>

    

int nulls=0;
float TotalMain=0;
// WiFi Parameters
const char* ssid = "SSID"; 
const char* password = "PASS"; 
const int httpsPort = 443; 
const char* host = "HOST"; 
const String url = ""; 
const char* fingerprint = "X";

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
 Serial.println(); 
 Serial.print("Connecting "); 
 while (WiFi.status() != WL_CONNECTED) { 
   delay(500); 
   Serial.print("."); 
 } 
 Serial.println(); 
 Serial.print("Connected, IP address: "); 
 Serial.println(WiFi.localIP()); 
}

float ConnectAndCalculate(int page) {
  const size_t bufferSize = JSON_ARRAY_SIZE(20) + 2*JSON_OBJECT_SIZE(2) + 20*JSON_OBJECT_SIZE(17);
  DynamicJsonBuffer jsonBuffer(bufferSize);
  jsonBuffer.clear();
  float Total=0;
  if (WiFi.status() == WL_CONNECTED) {
    WiFiClientSecure client; 
  if (!client.connect(host, httpsPort)) { 
   String wifistatus = "WIFI FAIL"; 
   Serial.println("connection failed"); 
   return 0; 
  }
   if (client.verify(fingerprint, host)) { 
    Serial.println(host);
    Serial.println("\n");
    //Serial.println("certificate matches"); 
   } else { 
    //Serial.println("certificate doesn't match"); 
  } 
   client.print(String("GET ") + "/a=" + page + " HTTP/1.1\r\n" + 
              "Host: " + host + "\r\n" + 
              "User-Agent: BuildFailureDetectorESP8266\r\n" + 
              "Connection: close\r\n\r\n"); 
    //Serial.println("request sent"); 

  while (client.connected()) { 

   String line = client.readStringUntil('\n'); 
   if (line == "\r") { 
     //Serial.println("headers received"); 
     
     break; 
   } 
  } 
      // JSON 
      String line = client.readStringUntil('\n'); 
      //Serial.print("My page: ");
      Serial.println(page);
      // Parsing

      int i=0;
      //Serial.println("######################");
      //Serial.println(line);
      //Serial.println("######################");
      jsonBuffer.clear();
      JsonObject& root = jsonBuffer.parseObject(line);
      JsonArray& result_orders = root["result"]["orders"];
      
      
      // Parameters
      for(i=0;i<20;i++){
        if (result_orders[i] == NULL) {
          Serial.print("Added NULL because: ");
          Serial.println(i);
          nulls++;
          break;
          }
        //Serial.println(result_orders[i]["orderStatus"].asString());
        if(result_orders[i]["orderStatus"]=="Completed Payments") {
        Total = Total + (float)result_orders[i]["estimatedCommission"];
        Serial.print("Order: ");
        Serial.println(result_orders[i]["estimatedCommission"].asString());
        }
      }
      //Serial.print("Total NULL: ");
      //Serial.println(nulls);
      client.stop();
      //if (nulls==0) 
      //  Total = Total + ConnectAndCalculate(++page);
      Serial.print("Total:");
      Serial.println((String)Total);

      return Total;
   
    }

  }
  

void loop() {
  // Check WiFi Status
  int page=0;
  while(nulls==0)
    TotalMain = TotalMain + ConnectAndCalculate(++page);
  Serial.println("Total:");
  Serial.println((String)TotalMain);
  nulls = 0;
  TotalMain = 0;
  // Delay
  Serial.println("I'm Done");
  delay(60000);
}`

bblanchon, 2018-10-25:
You're welcome Mat.
Thank you for using ArduinoJson.
If you like this project, please [cast a star ⭐️](https://github.com/bblanchon/ArduinoJson/stargazers)

bblanchon, 2018-10-26:
@mTny, your usage of `NULL` inspired me to write an article: [Why C++ programmers don't use NULL?](https://cpp4arduino.com/2018/10/26/why-cpp-programmers-dont-use-null.html)