bblanchon/ArduinoJson #836: JsonPair::value does not have class type

martinloren, 2018-10-26:
Tring to run the example for V6 Beta:

```
char json[] = "{\"first\":\"hello\",\"second\":\"world\"}";
DynamicJsonDocument doc;
deserializeJson(doc, json);
JsonObject root = doc.as<JsonObject>();

// using C++11 syntax (preferred):
for (auto kv : root) {
    Serial.println(kv.key.c_str());
    Serial.println(kv.value.as<char*>());
}
```

Got:

> error: invalid use of non-static member function
>        Serial.println(kv.key);
> 
> error: 'kv.ArduinoJson650_0_0::JsonPair::value' does not have class type
>        Serial.println(kv.value.as<char*>());
>
> error:  expected primary-expression before 'char'
>        Serial.println(kv.value.as<char*>());
>

martinloren, 2018-10-26:
Documentation need to be updated since **key** and **value** are private now.
Code become:
`Serial.print(kv.key());
Serial.print(": ");
Serial.println(kv.value().as<char*>());`

bblanchon, 2018-10-26:
Sorry about that, I'll update the documentation ASAP

closed 2018-10-27 by martinloren