bblanchon/ArduinoJson #839: Naming conflict

martinloren, 2018-10-29:
Please change:

`template <size_t CAPACITY>`

into 

`template <size_t JSON_CAPACITY>`

(or other else) and rename all CAPACITY to JSON_CAPACITY. The issue is that CAPACITY is too generic and used in other Frameworks that conflict with this library. Thanks!

bblanchon, 2018-10-29:
Hi @martinloren,

If I follow this logic, I need to prefix all names with `JSON_`:

* namespaces
* class names
* typenames
* template parameter names
* variable names
* function parameters names
* macros
* macro parameters names

That doesn't make sense.

Please address this request to STM32 HAL Library because, as soon as they define macros, it's up to them to define global unique names.

Thank you for your understanding Martin.

Regards,
Benoit

bblanchon, 2019-03-01:
Fixed in [version 6.9.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.9.1)

closed 2019-03-01 by bblanchon