bblanchon/ArduinoJson #840: Missing elements during serialization

dmnc-net, 2018-10-30:
Hi,
I'm using dynamic allocation and I've used the ArduinoJson Assistant which helped me with syntax and the elements count :) ... which is 14.

```
#define JSON_ELEMENTS               14
const size_t bufferSize = JSON_OBJECT_SIZE(JSON_ELEMENTS);
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonObject& json = jsonBuffer.createObject();
```

I'm doing some GPS tracker application and sending JSON with GPS data. I found a several elements missing so probably there is not enough of memory available during the process of adding elements.
But if I put some "fake" static data to the object before (during the `setup()`), everything is working good and later my `json` object contains all 14 elements, I have still some memory available (385B if I can trust [this library](https://github.com/McNeight/MemoryFree)):

```
  json["ccid"] = "1234567890123456789";
  json["imei"] = "123456789012345";
  json["timestamp"] = "12345678901234.000";
  json["lat"] = 11.00001;
  json["lon"] = 11.00001;
  json["alt"] = 123.4;
  json["speed"] = 0.1;
  json["course"] = 123.4;
  json["usat"] = 11;
  json["vsat"] = 11;
  json["cnr"] = 11;
  json["hdop"] = 1.1;
  json["pdop"] = 1.1;
  json["vdop"] = 1.1;
```

I found out that I can't add `json["element"] =` or `json.set("element", value)` in case of char arrays `char ccid[20]`, `char imei[16]`, `char timestamp[19]`. Problem occurs even with some test `char test[2] = "1"` but no problem with `char test = '1'` 

When I'm trying to isolate the code, everything is working good:
```
#include <ArduinoJson.h>
const size_t bufferSize = JSON_OBJECT_SIZE(14);
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonObject& json_test = jsonBuffer.createObject();

void setup() {
  Serial.begin(38400);
  char ccid[] = "1234567890123456789";
  char imei[] = "123456789012345";
  char timestamp[] = "12345678901234.000";
  json_test["ccid"] = ccid;
  json_test["imei"] = imei;
  json_test["timestamp"] = timestamp;
  json_test["lat"] = 11.00001;
  json_test["lon"] = 11.00001;
  json_test["alt"] = 123.4;
  json_test["speed"] = 0.1;
  json_test["course"] = 123.4;
  json_test["usat"] = 11;
  json_test["vsat"] = 11;
  json_test["cnr"] = 11;
  json_test["hdop"] = 1.1;
  json_test["pdop"] = 1.1;
  json_test["vdop"] = 1.1;
  Serial.print(F("Elements: "));
  Serial.println(json_test.size());
  json_test.printTo(Serial);
}

void loop() {
}
```

I don't know how AJ5 is working with char arrays but I think there must be a more efficient way how to reserve space in the heap than my trick with static data in the `setup()`. Many thank for some advice.

Env: Arduino UNO, ArduinoJSON v5.13.3

bblanchon, 2018-10-31:
Hi @dmnc-net,

ArduinoJson stores a string differently depending on its type:

* it stores `const char*` by pointer
* it stores `char*` and `String` by copy.

In your case, I can see three `char*` that ArduinoJson duplicates in the `JsonBuffer`.
That means that the `JsonBuffer` is too small, which explains the missing parts.

Please read:

* [Why are some parts missing?](https://arduinojson.org/v5/faq/why-some-parts-are-missing/?utm_source=github&utm_medium=issues)
* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)
* [How to reduce memory usage?](https://arduinojson.org/v5/faq/how-to-reduce-memory-usage/?utm_source=github&utm_medium=issues)

Regards,
Benoit

dmnc-net, 2018-11-01:
Hi, thanks for this clarification. I'm completely avoiding the `String`s but all that `char` arrays and pointers aren't my strong suite, I'll follow this issue.

I saw those links before I've asked. Now I'm reading through again and again and I have to consider how to get rid of global JsonBuffer (maybe my "filling" methods can return new object or work with the pointer to the object).

And of course I'd love to use static buffer but I thought it will be a waste of memory in some cases (when data are smaller - ie. speed 0.1 instead of 120.8)

And the duplication is probably big problem too, I'll try to rework the GPS data parser and class variables which I'm using for final JSON.

I'll back with the update and/or additional questions. Thank you.

dmnc-net, 2018-11-01:
One thing I still don't understand. When I prepare and fill json object with fake initialization data in exact length, why I can't rewrite them later with `json.set()`, the result contains still initialization data.

bblanchon, 2018-11-03:
Hi @dmnc-net,

I'm glad you finally found a solution.

By the way, the following sentence made me smile (in a good way of course):

> I'd love to use static buffer but I thought it will be a waste of memory in some cases (when data are smaller - ie. speed 0.1 instead of 120.8).

If I'm not mistaken, your program must be able to parse both `0.1` and `120.8`, so you need to allocate the two extra bytes anyway.

Your reasoning makes sense when you develop an application that runs on a computer because you want to be a good citizen and use as little memory as possible. However, when your program is the only one running on the machine, you should use as much memory as you need: don't be stingy! In particular, you need to make sure that the `JsonBuffer` is big enough for any valid input and the best way to do that is to use fixed memory allocation.

I hope that makes sense ;-)

Regards,
Benoit

closed 2018-11-03 by bblanchon