bblanchon/ArduinoJson #841: Json parsing problem

kuzmanzs, 2018-11-02:
Hi,

I have esp8266 with Arduino IDE.

I try to parse big json. Your calculator says it needs 7800 bytes.

I am using DynamicJsonBuffer jsonBuffer(3000).
If I good understood 3000 only initial size is and it can grow.

With basic program features before parsing I have 15520 free heap. And it works. It can grow up to 8080 bytes.

```
getting data
parsing response, freeMemory=15520
JsonBufferSize:0
parsing json..
JsonBufferSize:8080
parsing done,freeMemory=6760
```0

When I adding other memory consuming features lets say I have 14400 free bytes. 
```
getting data
parsing response, freeMemory=14400
JsonBufferSize:0
jsonBuffer.parseObject() failed
JsonBufferSize:3000
failed, freeMemory=11656
```
Ib that case memory could not grow up to 8080 bytes. I do not understand why. There are lot of free memory available. I have tested with same json input.

bblanchon, 2018-11-04:
Hi @kuzmanzs,

If I understand your question, you're asking why the `DynamicJsonBuffer` doesn't grow even if there is enough free memory.

It doesn't grow because it cannot allocate a block of 6KB. True, there are 11KB of free memory, but there is not a contiguous block of 6KB. Indeed, the free memory is composed of disjoint fragments, a phenomenom named [heap fragmentation](https://en.wikipedia.org/wiki/Fragmentation_(computing)).

ArduinoJson helps reducing the heap fragmentation because it allocates only one big chunk of memory. However, to leverage this feature, you have to specify the right size to the constructor of `DynamicJsonBuffer`, so it only makes one allocation. If the [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/) says 7800 bytes, then pass at least 8000 to the constructor; you can put more, but you should never pass less.

If you want to reduce heap fragmentation, you need to avoid dynamic memory allocation in the rest of the program. In particular, get rid of as much `String` instance as you can, they are the heap killers.

Regards,
Benoit

kuzmanzs, 2018-11-06:
Hi,

thanks your replay.

Could you help me, how can I Stream(ify) fs::File?

bblanchon, 2018-11-07:
Hi @kuzmanzs,

`File` is a `Stream`, so you can call `myFile.find()`.

Regards,
Benoit

kuzmanzs, 2018-11-08:
Hi,

oh, yes. Thanks :)

closed 2018-11-08 by kuzmanzs