bblanchon/ArduinoJson #844: Compiling-issues using Sloeber

MorpheusDe97, 2018-11-07:
Allthough I read about the existing problems with the ArduinoJson-Library and the Sloeber-IDE, there might be still some issues that are not being documented yet. 

I tried out some older and some newer versions of the major version 5, tried out all the three existing methods to install the lib (cloning the library into the lib-folder, using the header-file alone and using the Library-Manager,  and always got some compile errors using the lib. 

Most of the errors lead to the library itself, but as the error-outpur is quite verbose, it is really difficult to debug for me and to locate the errors. 

E.g. 

`error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<ArduinoJson::StaticJsonBuffer<500u>, void>'`

`error: no matching function for call to 'ArduinoJson::JsonVariant::as() const'`

` error: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, std::__cxx11::basic_string<char> >'`

` error: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, ArduinoJson::JsonObject&>'`

Actually, it is not the first time that I got issues using the ArduinoJson-Library with the Sloeber-IDE, but normally reinstalling the library would solve the problem. Now, unfortunately,  it doesn't seem to solve it. 

It would be great, if someone could help me.

bblanchon, 2018-11-07:
Hi @MorpheusDe97,

Your issue doesn't seem to be related to the Sloeber-IDE; I think there is a problem in your code.
Can you share the complete compiler output and the line in your program that causes this error?

Regards,
Benoit

MorpheusDe97, 2018-11-07:
Hi Benoit, 
thank you for your fast reply. 
As it is a really big error output, can I send it to you via E-mail?

MorpheusDe97, 2018-11-07:
Okay, it was indeed a mistake in my code, i used the wrong datatype! I was just a bit misslead by the errors in the ArduinoJson-Library!

closed 2018-11-07 by MorpheusDe97