bblanchon/ArduinoJson #845: [v6] Apparent Improper OK for "add":}

jeffsf, 2018-11-07:
At `commit e842838a2362fa3ca7b5c30c397ebfc68f1fabfc (HEAD -> 6.x, origin/6.x)` of 2018-10-25

Input of
```
"add":}
```
provides unexpected output indicating successful parsing
```
Utilization: 4
                       "add":} -> Ok
```
Expected `IncompleteInput` error
```
// ArduinoJson - arduinojson.org
// Portions Copyright Benoit Blanchon 2014-2018
// MIT License

#include <iostream>
#include <iomanip>
#include <cstring>
#include <string>

#include "ArduinoJson.h"

int main()
{
    std::string str;

    do {
        getline(std::cin, str);

        DynamicJsonDocument doc;

        DeserializationError derr = deserializeJson(doc, str.c_str());
        std::cout << "Utilization: " << doc.memoryUsage() << std::endl;

        std::cout << std::setw(30) << str << " -> " << derr << std::endl;

    } while ( str.length() );

    return 0;
}
```

macOS "High Sierra" 10.13.6, XCode, CLion using "C++ Compiler: Detected:"
```
$ c++ --version
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jeffsf, 2018-11-07:
Slightly different behavior with passing in a C-style string directly
```
|"add":}|
Utilization: 0
                           add -> Ok

Process finished with exit code 0
```
```
// ArduinoJson - arduinojson.org
// Portions Copyright Benoit Blanchon 2014-2018
// MIT License
//
// This example shows how to deserialize a JSON document with ArduinoJson.

#include <iostream>
#include <iomanip>

#include "ArduinoJson.h"

int main()
{
    char cstr[]{"\"add\":}"};

    std::cout << "|" << cstr << "|" << std::endl;

    DynamicJsonDocument doc;

    DeserializationError derr = deserializeJson(doc, cstr);
    std::cout << "Utilization: " << doc.memoryUsage() << std::endl;

    std::cout << std::setw(30) << cstr << " -> " << derr << std::endl;

    return derr;
}
```

bblanchon, 2018-11-08:
Hi @jeffsf,

It's not a bug it's a feature 😄 

[`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/?utm_source=github&utm_medium=issues) deserializes the first valid value from the input and stops.
A valid value can be:

* an object `{"add":1}`
* an array `["add"]`
* a string`"add"`
* a boolean `true`
* a number `42`, `3.14`.

[Mandatory demo](https://wandbox.org/permlink/9pz36zehUA7WwCuo)

Why stop parsing instead of reading the complete input?
Because this behavior is super useful for streams.

For example, if you have a multiline text file, with each line being a JSON object.
Even if the complete file is not a valid JSON document, each line is, you can repeatedly call `deserializeJson()` to read the file line by line.

Another example is when you receive a huge response from a web server.
No need to parse the complete response, you can jump to the location of interest and only parse the part you need.

Regards,
Benoit

jeffsf, 2018-11-08:
Love features, especially once I understand them and their potential impact on code behavior. I can definitely see the use case. 

Only 211 pages into _Mastering ArduinoJson_ so far, but will "play around" with reading directly from an input stream in a sandbox here. Especially knowing that I've got to size my current input buffer for not only the input, but also '\0's inserted after every "string-able" set of characters, I'm thinking it might make more sense just to read-parse directly into a StaticJsonDocument directly. That v6 can distinguish between "bad" JSON and OOM conditions gives me hope that I don't have to wrap the input stream too heavily.

I'll check back in here once I've explored the possibilities of direct reading.

What's the best way to get book comments/suggestions back to you for the v6 version?

bblanchon, 2018-11-09:
Hi @jeffsf,

Readers usually send me feedback via email.
You'll find the address in the book or on arduinojson.org.

Regards,
Benoit

closed 2018-11-09 by bblanchon

jeffsf, 2018-11-09:
If anyone else reads this, purchasing _Mastering ArduinoJson_ was well worth its purchase price in both development time and bytes saved.

Examining `ArduinoStreamReader` convinced me that `deserializeJson()` will "wait" for the Arduino Serial to time out before stopping parsing from `Serial`. As a result, I was able to remove "double buffering" of the serial input, as well as the unreliable complexity of declaring sizes of both the serial buffer and the `StaticJsonBuffer`. This assumes that both the Arduino and the sender respect RTS/CTS/DTR/DSR signaling, as well as not having any "long pauses" (process swapping, GC, Python GIL, ...) relative to the Arduino's serial timeout (1000 ms, by default).

A big mess of code around double buffering and skipping "leading garbage on the line" until `{` was seen now reduces to
```
        while ( not Serial.available() )
            continue;

        SerialApiV1::processLine(Serial, sensorSet);  // which calls deserializeJson(doc, Serial)
```
(skipping is handled in `skipSpacesAndComments()` in `JsonDeserializer`)

Now to refactor that method's name into something more appropriate!