bblanchon/ArduinoJson #846: [v6][Enhancement] DeserializationError is not switch() Friendly

jeffsf, 2018-11-08:
Using a `DeserializationError` as a switch condition results in compiler warnings that the "switch condition has boolean value" and subsequent reduction to "odds/evens" of the cases. Looking at the class definition, I am guessing that at least `clang` selects `operator bool()` rather than the `Code` as I had expected.

Work-around appears to be a functionally equivalent `if` tree.

To answer the "obvious" line of questions around the necessity or purpose for the code segment below, I'm fighting for bytes of RAM in an AVR device and put all my strings into program memory. The `UnifiedErrors` class lets me manage all error reporting in one place, with the related strings stored in program memory. The 65 bytes of strings in `DeserializationError::c_str()` are "huge" compared to the free RAM I have available.

```
        DeserializationError derr = deserializeJson(doc, str.c_str());

        if ( derr ) {
            switch ( derr ) {
                case DeserializationError::IncompleteInput :
                    parsedDto.error = UnifiedErrors::kJsonParseIncompleteInput;
                    break;
                case DeserializationError::InvalidInput :
                    parsedDto.error = UnifiedErrors::kJsonParseInvalidInput;
                    break;
                case DeserializationError::NoMemory :
                    parsedDto.error = UnifiedErrors::kJsonParseNoMemory;
                    break;
                case DeserializationError::NotSupported :
                    parsedDto.error = UnifiedErrors::kJsonParseNotSupported;
                    break;
                case DeserializationError::TooDeep :
                    parsedDto.error = UnifiedErrors::kJsonParseTooDeep;
                    break;
            }
```

bblanchon, 2018-11-13:
Hi @jeffsf,

I was not able to implement:

```c++
switch( derr ) {
  // ...
}
```

because it conflicts with the conversion to `bool`.

So, you need to call `.code()`:

```c++
switch( derr.code() ) {
  // ...
}
```

I also implemented the safe bool idiom to get a better error message in case you forget to append `.code()`.

Regards,
Benoit

closed 2018-11-13 by bblanchon

bblanchon, 2020-10-19:
[ArduinoJson 6.17.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.0) adds `DeserializationError::f_str()` which returns a Flash string.

commit 0a97d4c8 (2018-11-13): Added `DeserializationError::code()` to be used in switch statements (closes #846)