bblanchon/ArduinoJson #849: Json library: parseobject() failed error

magnumpro, 2018-11-20:
I was controlling LED from server. 
Firstly I was working with www.000webhost.com, it worked fine and I didn't get any error. I controlled LED from remote server. But only the issue was less speed. 

So I purchased new server from www.hostinger.com to overcome less speed. But there I m getting parseobject failed error in arduino code.

The same arduino code worked for free server (www.000webhost.com) and its not working for paid server (www.hostinger.com), even i tried with godaddy, there also i m getting the same error.

Any one please in this issue.

String line = client.readStringUntil('\r');
   //Serial.print(line);
   // we’ll parse the HTML body here
   if (section=="header") { // headers..
    
     if (line=="\n") { // skips the empty space at the beginning 
       section="json";
     }
   }
   else if (section=="json") {  // print the good stuff
     section="ignore";
     String result = line.substring(1);

     // Parse JSON
     int size = result.length() + 1;
     char json[size];
     result.toCharArray(json, size);
     StaticJsonBuffer<300> jsonBuffer;
     JsonObject& json_parsed = jsonBuffer.parseObject(json);

    [b]if (!json_parsed.success())
      {
        Serial.println("parseObject() failed"); // here my code stucked
        return;
      }[/b]

     String led = json_parsed["led"][0]["status"];

     if(count == 1){
       if(led == "on"){
         digitalWrite(D1, 1);
         delay(100);
         Serial.println("D1 is On..!");
       }
       else if(led == "off"){
         digitalWrite(D1, 0);
         delay(100);
         Serial.println("D1 is Off..!");
       }
       count = 0;
     }

   }

bblanchon, 2018-11-20:
Hi @magnumpro,

All the reasons for a parsing failure are described in [Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/?utm_source=github&utm_medium=issues).

In your case, I believe that the [Chunked Transfer Encoding](https://fr.wikipedia.org/wiki/Chunked_transfer_encoding) is causing the problem.
Did you specify `HTTP/1.0` in the request?

Also, your program uses the `String` class excessively, so you're likely to have a [memory fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) issue. Please check out the example [JsonHttpClient.ino](https://arduinojson.org/v5/example/http-client/?utm_source=github&utm_medium=issues) to see how to implement an efficient HTTP client with ArduinoJson.

Regards,
Benoit

magnumpro, 2018-11-24:
Hi bblanchon,
 
   I tried with this "HTTP/1.0" and it worked fine. Thank you so much.