bblanchon/ArduinoJson #852: query string

fgrs, 2018-11-23:
is there a way of creating query string out of flat json using this library?

bblanchon, 2018-11-24:
Hi @fgrs,

I'm assuming that by "query string" you're referring to something like:

    pi=3.14&answer=42

The answer is no; ArduinoJson doesn't include such feature.

However, you can easily write a function to do that:

```c++
// for ArduinoJson 5
String makeQueryString(const JsonObject& obj) {
  String result;
  bool needSeparator = false;
  for (const JsonPair& p : obj) {
    if (needSeparator) result += "&";
    result += p.key;
    result += "=";
    p.value.printTo(result);
    needSeparator = true;
  }
  return result;
}
```

See online demo:
https://wandbox.org/permlink/uHInbgSDVsLpenv0

Regards,
Benoit

bblanchon, 2018-11-29:
Hi @fgrs,

I'd like to know if this function is useful, so as to put it on arduinojson.org.
Did you try it?

Regards,
Benoit

closed 2018-12-19 by bblanchon