bblanchon/ArduinoJson #855: Obj.printTo SPIFFS  File   -->  ESP32 reboot

spencer1979, 2018-11-29:
Here is my code 
bool UpdateRsense(int ID, float rsense)
{
  String jsonStr;
  StaticJsonBuffer<512> readBuffer; //create Json buffer
  // Parse the root objec

  unique_ptr<char> config1 = ReadData(ConfigFilePath);
  if (config1 == nullptr)
  {
    Serial.println(F("NullPtr return "));
    return false;
  }
  JsonObject &root = readBuffer.parseObject(config1.get());
  if (!root.success())
  {
    Serial.println(F("Failed to read file, using default configuration"));
    return false;
  }
  const char *path = root["FILE"][ID]; ///get File path
  config1.reset();
  readBuffer.clear(); //clear json buffer
  config1 = ReadData(path);

  if (config1 == nullptr)
  {
    Serial.println(F("Nullptr return "));
    return false;
  }

  JsonObject &obj = readBuffer.parseObject(config1.get());

  if (!obj.success())
  {
    Serial.printf("Parse object fail path:%s\r\n", path);
    return false;
  }
  float num = floor(1000 * rsense) / 1000;
  if (!root.containsKey("RSENSE"))
  {
    Serial.printf("Can't find the key :%s\r\n", rsense);
    return false;
  }
  obj["RSENSE"] = num; //update RSENSE
  if (!SPIFFS.exists(path))
  {
    Serial.printf("File open fail :%s\r\n", path);
    return false;
  }

  File updateConfig = SPIFFS.open(path, "w");

  if (!updateConfig)
  {
    Serial.printf("File open fail :%S\r\n", path);
    return false;
  }
  obj.printTo( updateConfig );
 
  return true;
}

Exception decode  
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060d30  A0      : 0x800f3efc  A1      : 0x3ffbd310
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x3f410ec3  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x20000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400014fd:0x3ffbd310 0x400f3ef9:0x3ffbd320 0x400fb44a:0x3ffbd630 0x400fb486:0x3ffbd6c0 0x400e048e:0x3ffbd700 0x40132cdb:0x3ffbd760 0x400d6be0:0x3ffbd7e0 0x400d429e:0x3ffbd810 0x400d4952:0x3ffbd850 0x400d4c64:0x3ffbdad0 0x400d4e54:0x3ffbdb10 0x4014fa93:0x3ffbdb30 0x4015029a:0x3ffbdb50 0x400d7f6d:0x3ffbdba0 0x400d7f89:0x3ffbdbe0 0x40131d51:0x3ffbdc00 0x40131f7d:0x3ffbdc50

Decoding stack results
0x400f3ef9: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x400fb44a: _vsnprintf_r at ../../../.././newlib/libc/stdio/vsnprintf.c line 72
0x400fb486: vsnprintf at ../../../.././newlib/libc/stdio/vsnprintf.c line 41

 my platform :ESP32 Nodemcu 32S

bblanchon, 2018-11-29:
Hi @spencer1979,

The stacktrace doesn't point to ArduinoJson, what makes you think the library is guilty?

To me, it looks like there is something wrong with one of the `printf()`s.
The following one is suspicious:

```c++
Serial.printf("File open fail :%S\r\n", path);
```

I never head of a `%S`; did you mean `%s`?

Regards,
Benoit

spencer1979, 2018-11-29:
Hi @bblanchon ,  thanks ..

closed 2018-11-29 by spencer1979