bblanchon/ArduinoJson #856: 'class ArduinoJson660_0_0::StaticJsonDocument<200u>' has no member named 'parseObject'

TomAvecLeVdsl, 2018-11-29:
```cpp
void loop() {
  if (yeelight->feedback()) {
    Serial.print("device: ");
    Serial.println(yeelight->getLocation());

    if (!yeelight->isPowered()) {
      digitalWrite(LED, HIGH);
      Serial.println(yeelight->sendCommand("set_power", "[\"on\", \"smooth\", 500]"));
      // Serial.println(yeelight->sendCommand("set_name", "[\"tv\"]"));
    }

    JsonObject& root = JsonDocument.parseObject(yeelight->sendCommand("get_prop", "[\"power\", \"name\"]")); // stucked here 
    const char* state = root["result"][0];
    const char* name = root["result"][1];
    Serial.print("- power is: ");
    Serial.println(state);
    Serial.print("- name is: ");
    Serial.println(name);
  }
}

`
Logs:
`C:\Users\Tom\Desktop\yeelight-arduino-master\yeelight_remote\yeelight_remote.ino: In function 'void loop()':

yeelight_remote:34: error: 'class ArduinoJson660_0_0::StaticJsonDocument<200u>' has no member named 'parseObject'

     JsonObject& root = JsonDocument.parseObject(yeelight->sendCommand("get_prop", "[\"power\", \"name\"]"));

                                     ^

exit status 1
'class ArduinoJson660_0_0::StaticJsonDocument<200u>' has no member named 'parseObject'`

Thanks for replying :)

bblanchon, 2018-11-30:
Hi @TomAvecLeVdsl,

Please check out the examples:

* [Examples for ArduinoJson version 5](https://arduinojson.org/v5/example/parser/?utm_source=github&utm_medium=issues)
* [Examples for ArduinoJson version 6](https://arduinojson.org/v6/example/parser/?utm_source=github&utm_medium=issues)

I recommend version 5 because 6 is still in beta, so you may need to downgrade.

Regards,
Benoit

TomAvecLeVdsl, 2018-11-30:
ok , thanks :)

closed 2018-11-30 by TomAvecLeVdsl