bblanchon/ArduinoJson #857: crash on arduino ESP8266

manugre, 2018-12-03:


manugre, 2018-12-03:
I'm trying to adapt this code to run on thingsboard 👍 
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L1626
I need to deserialize json rpc request 
I change to this : 

void receivingMQTT(String const topic_name, String const data_str) {
  char* tok_ptr;
  uint64_t code = 0;
  uint16_t nbits = 0;
  uint16_t repeat = 0;

  debug("Receiving data by MQTT topic :" + topic_name);

  // Make a copy of the callback string as strtok destroys it.
  debug("MQTT Payload (raw): " + data_str);
  
  DynamicJsonBuffer jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(data_str);
  String const callback_str = root["ircode"];
  char* callback_c_str=strdup(callback_str.c_str());
  
... all over is unchanged

Then compil & upload on ESP8266 (ESP01)
when i send a rpc mqtt request, ESP crash 👍 
17:53:49.744 -> 0002609.389: MQTT Payload (raw): {"method":"sendIR","params":"1500"}
17:53:49.744 -> Exception (28):
17:53:49.744 -> epc1=0x4021c572 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
17:53:49.744 -> 
17:53:49.744 -> ctx: cont 
17:53:49.744 -> sp: 3ffffbe0 end: 3fffffd0 offset: 01a0
17:53:49.777 -> 
17:53:49.777 -> >>>stack>>>
...

If i change the 3 "json lines to this, no crash (but no work as i want, of course 👍 ) : 

 //JsonObject& root = jsonBuffer.parseObject(data_str);
  //String const callback_str = root["ircode"];
  //char* callback_c_str=strdup(callback_str.c_str());

  String const callback_str ="XXXX";
  char* callback_c_str=strdup(callback_str.c_str());

bblanchon, 2018-12-04:
Hi @manugre,

I don't think the problem comes from ArduinoJson, but more likely of what the program does with `callback_c_str`.

Can you try to get more information with [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) or [ArduinoTrace](https://github.com/bblanchon/ArduinoTrace)?

I see you're using a lot of `String`s, so it's quite possible that you're having a [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) problem.
I recommend reading [this article](https://cpp4arduino.com/2018/11/21/eight-tips-to-use-the-string-class-efficiently.html) to reduce the heap fragmentation caused by strings.

BTW, you can probably simplify the code above to:

```c++
const char* callback_c_str = root["ircode"];
```

With this code, `callback_c_str` is null if `ircode` is missing, and the memory belongs to the JsonBuffer.

Regards,
Benoit