bblanchon/ArduinoJson #859: Compilation failure with STM32F103C8T6 Blue pill

shaun-dm-stewart, 2018-12-04:
Hi,
Arduino IDE 1.8.7 and Visual Micro VS2017
ArduinoJson library 6 Beta

I have attempted to compile an example sketch against an STM32F103C8T6 Blue Pill board.  The compiler returns an error to the effect that  strncmp_P cannot be found.  I checked the pgmspace.h file for this board and indeed there is no such macro.  I have temporarily added the macro
#define strncmp_P(a, b, n) strncmp((a), (b), (n)) and the code compiles but that is a hack to core library files.  Is there a workaround or alternative solution for this?

regards

Shaun

bblanchon, 2018-12-04:
Hi Shaun,

You can disable the Flash string support in the library.
Simply define [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/?utm_source=github&utm_medium=issues) to `0`.

However, this issue probably deserve a Pull Request to fix the core; which one are you using?

Regards,
Benoit

shaun-dm-stewart, 2018-12-04:
Hi,

Thanks for getting back to me so quickly.  That worked like a charm.  The core I am using is from Roger Clarke's Arduino_stm32 GitHub project.  The pgmspace.h file is located in
/STM32F1/cores/maple/avr/.  I am rather new to this so forgive the poor description.

Once again many thanks.  I really like the library by the way.

regards

Shaun

shaun-dm-stewart, 2018-12-04:
Sorry! closed by accident.  I am happy to close this incident though :)  Told you I was new to this

bblanchon, 2018-12-05:
Thanks for the update!
Don't hesitate to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) if you like this project.

closed 2018-12-05 by bblanchon

shaun-dm-stewart, 2018-12-05:
Star cast ??

Regards

Shaun

Closed #859<https://github.com/bblanchon/ArduinoJson/issues/859>.