bblanchon/ArduinoJson #860: deserializeJson changes original variable

Distorzion, 2018-12-04:
When I call deserialize on a char[400], the original value of the variable is getting changed. I'm guessing some sort of \0 being added to the origin variable. is this behavior normal? is there anyway to avoid it? Board is an ESP8266 and v6.
`
char _wtconf[400]; <-- Data is loaded from SPIFF

bool setWateringData(){
      Serial.println(&_wtconf[0]); <-- prints {"mtth":21,"mttm":25,"nsth":23,"nstm":27,"h":13,"l":18,"sd":0,"n":"Lechuga"}
      Serial.println( ESP.getFreeHeap());
      StaticJsonDocument<400> doc;
      auto err = deserializeJson(doc, _wtconf, 400);
      Serial.println(&_wtconf[0]); <-- prints only "mtth"
      JsonObject root = doc.as<JsonObject>();
      Serial.println( ESP.getFreeHeap());
      if(err){
        _debugger->error(FPSTR(sets_str_15), err.c_str());
        _waterDataExists = false;
        return _waterDataExists;
      }
      
      _waterDataExists = true;
      Serial.println(_mtth); <-- prints 0 (unset val)
      _mtth = root["mtth"];
      Serial.println(_mtth); <-- prints 21 (saved and read value)
      _mttm = root["mttm"];
      _nsth = root["nsth"];
      _nstm = root["nstm"];
      _sharing = root["sd"];
      root["n"].as<String>().toCharArray(_plant_name, 100);
      _humidity = root["h"];
      Serial.println(_water_lt);
      _water_lt = root["l"];
      Serial.println(_water_lt);

      return _waterDataExists;
    }
`

bblanchon, 2018-12-05:
Hi @Distorzion,

Please read:

* [Why is the input modified?](https://arduinojson.org/v6/faq/why-is-the-input-modified/?utm_source=github&utm_medium=issues)

I've just updated the article for version 6.

Regards,
Benoit

Distorzion, 2018-12-06:
Thanks Benoit, closing issue

closed 2018-12-06 by Distorzion

bblanchon, 2018-12-07:
You're welcome, thanks for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: