bblanchon/ArduinoJson #862: deserializeJson() raises IncompleteInput when passing end of line

scls19fr, 2018-12-06:
Hello,

The following message

    deserializeJson() failed: IncompleteInput

is shown on `Serial` monitor with

Receiver
```C
...
if (Serial1.available()) { 
    // Initialisation du format Json
    StaticJsonDocument <200> doc;
    DeserializationError error = deserializeJson(doc, Serial1);
    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      return;
    }
}
...
```

and sender

```C
StaticJsonDocument<200> doc; // Création d'un document
JsonObject root = doc.to<JsonObject>();

root["evt"] = 1; // Déclaration des différentes variables que contient le message
...

serializeJson(root, Serial1); // Envoi des données sur le module Xbee
Serial1.println();

// Serial.println(m_sector_last);
// Serial.println(m_sector);
serializeJsonPretty(root, Serial); // Envoi des données sur le moniteur Série
Serial.println();
```

I think this error is because of  `Serial1.println();` but I need this end of line to be send on the sender side because I have an other receiver (with a XBee connected directly to a computer and with a Go script) and so without Arduino and ArduinoJson.
This end of line help me to parse JSON message more easily with Go...

So I wonder if there's a work around to keep this EOL and without having this IncompleteInput message on Arduino receiver side.

Kind regards

bblanchon, 2018-12-07:
Hi @scls19fr,

ArduinoJson ignores spaces and line breaks.
See a demo here: https://wandbox.org/permlink/zUBeVMLYQK1g7F1Z

I think your program fails because of a timeout.
Please try to [increase the timeout value](https://www.arduino.cc/en/Serial/SetTimeout).

If you want to know where the parser stopped reading the stream, you can read the content of Serial1 just after `deserializeJson()` returns `IncompleteInput`.

Regards,
Benoit

scls19fr, 2018-12-07:
Thanks @bblanchon I will try this.

scls19fr, 2018-12-07:
Reopening because increasing timeout doesn't help.

JSON parsed message is displayed correctly but increasing timeout just delays moment when error message appears.

Adding 

    Serial.println(Serial1.readString());

when `error` occurs just displays "end of line"

So I think I need to filter out when just EOL is received.

Any idea?

bblanchon, 2018-12-07:
Indeed, if the Stream only contains a line break and nothing else, the parser considers it's an error.
See a demo: https://wandbox.org/permlink/wt4zx5TQAqhzuhmb

Are you in this situation?
If so, what do you expect from the library?

scls19fr, 2018-12-07:
What is unclear to me is that 2 messages are received when only one is send... one with the json string... and an other one with just EOL.I was expecting only one message (json string finishing with EOL... and probably \0 also)

Maybe serializing using serializeJson should provide an option to send eol (between json string and \0) and it should be handled on receiver / deserialiser side.

An other approach could be done on deserialiser side... if an empty message (or just an eol) is received maybe a specific error should be raised so we could catch and print error that are not empty message error.

scls19fr, 2018-12-09:
I've just try using XCTU to see received string... There is no 
`0x00` but `0x0D 0x0A` when I add `Serial1.println();`
Here is an example of received message:

`7B 22 65 76 74 22 3A 32 2C 22 69 64 22 3A 31 32 33 34 35 2C 22 6D 69 6C 22 3A 37 37 37 36 33 2C 22 6C 69 6E 65 22 3A 30 7D 0D 0A`

for

```
{"evt":2,"id":12345,"mil":77763,"line":0}

```

with `Serial1.println();` but such message is received as

`7B 22 65 76 74 22 3A 32 2C 22 69 64 22 3A 31 32 33 34 35 2C 22 6D 69 6C 22 3A 37 37 37 36 33 2C 22 6C 69 6E 65 22 3A 30 7D`

and a message with only `0D 0A`

Without `Serial1.println();` received message looks like

`7B 22 65 76 74 22 3A 32 2C 22 69 64 22 3A 31 32 33 34 35 2C 22 6D 69 6C 22 3A 37 37 37 36 33 2C 22 6C 69 6E 65 22 3A 30 7D`

What is necessary is that message with ending `0D 0A` should be send in one piece

bblanchon, 2018-12-09:
OK, I see. So your input is:

     {"hello":"world"}\r\n

You call `deserializeJson()`, so it reads the object but leave that behind:

    \r\n

Then you call `deserializeJson()` again, it ignores the spaces, waits for the timeout, and returns `IncompleteInput`.

There is a straightforward workaround: simply flush the trailing spaces and wait for new input before calling `deserializeJson()`. Here is what I did in the case-study "Recursive Analyzer" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues):

```c++
// Flushing Serial input buffer
while (Serial.available())
  Serial.read();

// Waiting for a JSON document in Serial
while (!Serial.available())
  delay(50);
```

In your case, you probably want to restrict the characters to only CR and LF. You can easily add such test by calling [Stream::peek()](https://www.arduino.cc/reference/en/language/functions/communication/stream/streampeek/) in the first loop condition.

closed 2018-12-30 by bblanchon