bblanchon/ArduinoJson #863: Callback mode

me21, 2018-12-07:
I need to parse JSON document which is an array with unknown number of items, and I thought if it is really necessary to store the whole document in memory.
Is it possible to provide callback mode to avoid input duplication at all?
For example, consider the following JSON structure:
```
[{"id": 1, "name": "test user"}]
```

If some callbacks are provided, then they would be called in the following order:
* arrayStart
* objectStart
* jsonPair("id", 1)
* jsonPair("name", "test user")
* objectEnd
* arrayEnd

It would allow to reduce memory requirements for really long JSON documents. If JSON is coming from Stream, it would never have to be wholly stored in memory.

For nested objects/arrays, the callbacks could look like
* jsonPair("object_name", ArduinoJson::nestedObject)
* objectStart
...
* objectEnd

If there is an error in JSON structure, i.e.
```
[{"id": 1]
```

the callbacks could look like
* arrayStart
* itemStart
* jsonPair("id", 1)
* jsonStructureError

This stream-like JSON parsing should relax memory requirements for large JSON documents, I think.

bblanchon, 2018-12-07:
Hi @me21,

You're in the wrong place :-)
The library you're looking for is [json-streaming-parser](https://github.com/squix78/json-streaming-parser).

To handle large arrays with ArduinoJson, you need to deserialize each element one after the other.
It's easier than it sounds; see [Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/v5/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

Regards,
Benoit

me21, 2018-12-07:
Thanks for both suggestions!

closed 2018-12-07 by me21