bblanchon/ArduinoJson #864: Sending Json object in POST request to nodeJS server

Zeni241, 2018-12-07:
Hi
I am trying to send Json object in POST request to nodeJS server using ArsuiniJson. My board is esp32.

I am getting following error:
 
```
19:23:23.933 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
19:23:23.933 -> Core 1 register dump:
19:23:23.933 -> PC      : 0x4015c29c  PS      : 0x00060630  A0      : 0x800d5e74  A1      : 0x3ffb1c80  
19:23:23.933 -> A2      : 0x3ffb1d28  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
19:23:23.968 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x73615000  A9      : 0x3ffb1c70  
19:23:23.968 -> A10     : 0x3ff87dd4  A11     : 0x3ffb9a38  A12     : 0x0000001f  A13     : 0x0000ff00  
19:23:23.968 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
19:23:23.968 -> EXCVADDR: 0x00000010  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffff9  
19:23:23.968 -> 
19:23:23.968 -> Backtrace: 0x4015c29c:0x3ffb1c80 0x400d5e71:0x3ffb1ca0 0x400d6e3d:0x3ffb1cc0 0x400d28e9:0x3ffb1cf0 0x400d2b8e:0x3ffb1f60 0x400d2bd1:0x3ffb1f80 0x4013f5db:0x3ffb1fa0
19:23:24.002 -> 
19:23:24.002 -> Rebooting...
19:23:24.002 -> ets Jun  8 2016 00:22:57
19:23:24.002 -> 
19:23:24.002 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:23:24.002 -> configsip: 0, SPIWP:0xee
19:23:24.002 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:23:24.002 -> mode:DIO, clock div:1
19:23:24.002 -> load:0x3fff0018,len:4
19:23:24.036 -> load:0x3fff001c,len:952
19:23:24.036 -> load:0x40078000,len:6084
19:23:24.036 -> load:0x40080000,len:7936
19:23:24.036 -> entry 0x40080310
```
My code in Arduino IDE is as follow:

  ```
const int capacity = JSON_OBJECT_SIZE(2);
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["ver"] = 2;
  root["id"] = "11181022000001";
 root.printTo(Serial);
  String databuf;
  root.printTo(databuf);

 http.begin("https://mydomain.com/valves" );        
  int httpCode = http.POST(databuf);

```
Any help please.

bblanchon, 2018-12-07:
Hi @Zeni241,

I don't see what caused the exception from the snippet above.

Can you try to decode the backtrace with [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)?
Or, can you try to add traces with [ArduinoTrace](https://github.com/bblanchon/ArduinoTrace) to identify the line that causes the exception?

Regards,
Benoit

Zeni241, 2018-12-08:
Hi bblanchon 
Thanks for your response. I solved the issue by pulling the json thing out of main code for posting data to server. Here is my new code which is working fine.

```
String ver="v2"; 
const char *username = "12345689";
String postMessage;

 void cookJson(){
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["ver"] = ver;
  root["id"] = username;
 root.printTo(Serial);
  root.printTo(postMessage);
 }
```
```
void checkserver(){

         HTTPClient http;
         http.begin("https://mydomain.com/valve" );      
         http.addHeader("Content-Type", "application/json");
         int httpCode = http.POST(postMessage);
         if(httpCode > 0) {

               My code etc. etc.

         }
     }
```

bblanchon, 2018-12-09:
Excellent!
Thanks for the update @Zeni241, and thanks for using ArduinoJson.

closed 2018-12-09 by bblanchon