bblanchon/ArduinoJson #865: DynamicJsonBuffer fails to parse

yolkhovyy, 2018-12-09:
Platform: Arduino ESP8266

DynamicJsonBuffer fails to parse the same json string after running some time. No crash, or lack of memory reported.

```
Received: thing/FFFFFFFFFFFF/evt/time/json:{"timestamp":1544350501,"tzoffset":7200}
Time is set: 1544350501
Some time later...
Received: thing/FFFFFFFFFFFF/evt/time/json:{"timestamp":1544350921,"tzoffset":7200}
Json parsing error in {"timestamp":1544350921,"tzoffset":7200}

```

The code:

```
DynamicJsonBuffer json;
...
System::printf("Received: %s:%s\n", topic.c_str(), payload.c_str());
  if (topic == Mqtt::topicEvtTime()) {
    JsonObject& root = json.parse(payload);
    if (!root.success()) {
      System::printf("Json parsing error in %s\n", payload.c_str());
    } else {
      // UTC timestamp, seconds
      if (root.containsKey(TIMESTAMP)) {
        setTime(root[TIMESTAMP].as<long>());
        System::printf("Time is set: %s\n", String(now()).c_str());
      }  
    }    
  }
...

```

yolkhovyy, 2018-12-09:
The payload is Arduino String.

bblanchon, 2018-12-09:
Hi @yolkhovyy, 

Supposing that the `DynamicJsonBuffer` is not global, I would say that it's a [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) problem.

Please read:

* [The first parsing succeeds, why do the next ones fail?](https://arduinojson.org/v5/faq/the-first-parsing-succeeds-why-do-the-next-ones-fail/)

Regards,
Benoit

yolkhovyy, 2018-12-09:
> Hi @yolkhovyy,
>
> Supposing that the `DynamicJsonBuffer` is not global, I would say that it's a [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) problem. [...]

The DynamicJsonBuffer is global, sorry did not make it clear.

yolkhovyy, 2018-12-09:
Could you please review this piece - it might be going wrong not in the code above but here (this runs periodically):

```
    JsonObject& root = json.createObject();
    root[TIMESTAMP] = now();

    if (!isnan(temperature)) {
      JsonObject& nested = root.createNestedObject(TEMPERATURE);
      nested[VALUE] = temperature;
      nested[UNITS] = String(celsius);
    }
    
    if (!isnan(humidity)) {
      JsonObject& nested = root.createNestedObject(HUMIDITY);
      nested[VALUE] = humidity;
      nested[UNITS] = "%";
    }
  
    if (!isnan(luminosity)) {
      JsonObject& nested = root.createNestedObject(LIGHT);
      nested[VALUE] = humidity;
      nested[UNITS] = "Lux";
    }
    
    String payload;
    root.printTo(payload);

    bool result = Mqtt::publish(Mqtt::topicEvtPubState(), payload);
    System::printf("Published %s:%s result %s\n", Mqtt::topicEvtPubState().c_str(), payload.c_str(), result ? "Ok" : "Failure");

```

yolkhovyy, 2018-12-09:
> I would say that it's a [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html)

It's not that easy to fragment heap, you need to do very special things. I think I probably have a leak.

bblanchon, 2018-12-10:
Hi @yolkhovyy,

> It's not that easy to fragment heap, you need to do very special things

I cannot disagree more. On the contrary, it's easy to fragment the heap, and you need to do special things to avoid it. And that's particularly true when you use the `String` class.

Calling `String::reserve()` can help to reduce the heap fragmentation, but **only if** you use a constant for the size.

That's why the best is often to get rid of the `String` class, like that:

```c++
char payload[1024];
root.printTo(payload);
```

Did you remove the global `JsonBuffer`?

See also:

* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/v5/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)
* [8 tips to use the String class efficiently](https://cpp4arduino.com/2018/11/21/eight-tips-to-use-the-string-class-efficiently.html)

Regards,
Benoit

yolkhovyy, 2018-12-11:
JsonBuffer in local scope works fine - good to know, thanks for your help!

closed 2018-12-11 by yolkhovyy

yolkhovyy, 2018-12-11:
> > It's not that easy to fragment heap, you need to do very special things
> 
> I cannot disagree more. On the contrary, it's easy to fragment the heap

Well, I'd say it's easy to screw heap, yes. Fragmenting, in normal circumstances - not sure. Malloc/free is an ingenious piece of software: https://reverseengineering.stackexchange.com/questions/15033/how-does-glibc-malloc-work

Not sure though which libc Arduino uses.

bblanchon, 2018-12-12:
On AVR, it uses [avr-lib](https://www.nongnu.org/avr-libc/)'s implementation of malloc() that is rather simplistic.
On ESP, it uses [umm_malloc()](https://github.com/rhempel/umm_malloc) which is a little more sophisticated but does produce a lot of fragmentation.

But even with a good implementation of `malloc()` you can have fragmentation, it's just that, on a PC, you don't feel the pain so quickly.