bblanchon/ArduinoJson #869: how to serializeJson to String in ver 6 ?

mlesn, 2018-12-17:
Hello,

How can I serialize json just to `String`? (not `Serial `etc)

My code:

```
  DynamicJsonDocument doc(53);
  JsonObject root  = doc.to<JsonObject>();
  root["entity_id"] = "test entity";

  serializeJson(root, Serial);
  USE_SERIAL.println();

  String json_string;
  serializeJson(root, json_string);

  USE_SERIAL.printf("[JSON] output json: %s\n", json_string);
```

Results with:

```
{"entity_id":"test entity"}
[JSON] output json: `⸮⸮?

```
My platform: ESP32 / Arduino IDE

Regards,
M

bblanchon, 2018-12-18:
Hi M,

The problem is in the `printf()` statement, it should be:

```c++
USE_SERIAL.printf("[JSON] output json: %s\n", json_string.c_str());
```

because `%s` expects a regular C string (ie a `const char*`), not a `String` object.

Regards,
Beniut

mlesn, 2018-12-19:
Hello,

Thank you for answer ;-)
It works... however it looks like `String` object generated by `serializeJson `results in another problems with conversions in my code.

I tought it will be better to use `const char*` as a results from json_serialization.

However my code doesn't compile when I try to use this type of variable:

```
const char* json_string;
serializeJson(root, json_string);
```

Is this possible to use `const char*` for serialization?

Regards, M

bblanchon, 2018-12-19:
Hi @mlesn,

It's a `const` pointer, so [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/?utm_source=github&utm_medium=issues) is not allowed to write to it.

Now, even if you remove `const`, you still need to initialize the pointer.

The best solution is to use an array instead:

```c++
char json_string[256];
serializeJson(root, json_string);
```

mlesn, 2018-12-19:
> ```c++
> char json_string[256];
> serializeJson(root, json_string);

Thank you for suggestion but it doesn't work. Compilation fails with following errors:

```
Arduino: 1.8.7 (Windows 7), Board: ""

In file included from C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10:0,

                 from C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson.hpp:25,

                 from C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Priv\Arduino.IDE\exercise\sketch_dec19a_JSON_test\sketch_dec19a_JSON_test.ino:1:

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp: In instantiation of 'void ArduinoJson670_0_0::JsonWriter<TWriter>::writeRaw(const char*) [with TWriter = char* [256]]':

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:82:5:   required from 'void ArduinoJson670_0_0::JsonSerializer<TWriter>::visitNull() [with TWriter = char* [256]]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectFunctions.hpp:16:5:   required from 'void ArduinoJson670_0_0::objectAccept(const ArduinoJson670_0_0::CollectionData*, Visitor&) [with Visitor = ArduinoJson670_0_0::JsonSerializer<char* [256]>]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectRef.hpp:22:17:   required from 'void ArduinoJson670_0_0::ObjectRefBase<TData>::accept(Visitor&) const [with Visitor = ArduinoJson670_0_0::JsonSerializer<char* [256]>; TData = ArduinoJson670_0_0::CollectionData]'

c:\priv\arduino.ide\libraries\arduinojson\src\arduinojson\serialization\serialize.hpp:21:3:   required from 'typename ArduinoJson670_0_0::enable_if<(! ArduinoJson670_0_0::IsWriteableString<TPrint>::value), unsigned int>::type ArduinoJson670_0_0::serialize(const TSource&, TPrint&) [with TSerializer = ArduinoJson670_0_0::JsonSerializer; TSource = ArduinoJson670_0_0::ObjectRef; TPrint = char* [256]; typename ArduinoJson670_0_0::enable_if<(! ArduinoJson670_0_0::IsWriteableString<TPrint>::value), unsigned int>::type = unsigned int]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:95:35:   required from 'size_t ArduinoJson670_0_0::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson670_0_0::ObjectRef; TDestination = char* [256]; size_t = unsigned int]'

C:\Priv\Arduino.IDE\exercise\sketch_dec19a_JSON_test\sketch_dec19a_JSON_test.ino:59:34:   required from here

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp:135:13: error: request for member 'write' in '((ArduinoJson670_0_0::JsonWriter<char* [256]>*)this)->ArduinoJson670_0_0::JsonWriter<char* [256]>::_writer', which is of non-class type 'char* [256]'

     _length += _writer.write(reinterpret_cast<const uint8_t *>(s), strlen(s));

             ^

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp: In instantiation of 'void ArduinoJson670_0_0::JsonWriter<TWriter>::writeRaw(char) [with TWriter = char* [256]]':

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp:34:13:   required from 'void ArduinoJson670_0_0::JsonWriter<TWriter>::beginObject() [with TWriter = char* [256]]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:41:5:   required from 'void ArduinoJson670_0_0::JsonSerializer<TWriter>::visitObject(const ArduinoJson670_0_0::CollectionData&) [with TWriter = char* [256]]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectFunctions.hpp:14:5:   required from 'void ArduinoJson670_0_0::objectAccept(const ArduinoJson670_0_0::CollectionData*, Visitor&) [with Visitor = ArduinoJson670_0_0::JsonSerializer<char* [256]>]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectRef.hpp:22:17:   required from 'void ArduinoJson670_0_0::ObjectRefBase<TData>::accept(Visitor&) const [with Visitor = ArduinoJson670_0_0::JsonSerializer<char* [256]>; TData = ArduinoJson670_0_0::CollectionData]'

c:\priv\arduino.ide\libraries\arduinojson\src\arduinojson\serialization\serialize.hpp:21:3:   required from 'typename ArduinoJson670_0_0::enable_if<(! ArduinoJson670_0_0::IsWriteableString<TPrint>::value), unsigned int>::type ArduinoJson670_0_0::serialize(const TSource&, TPrint&) [with TSerializer = ArduinoJson670_0_0::JsonSerializer; TSource = ArduinoJson670_0_0::ObjectRef; TPrint = char* [256]; typename ArduinoJson670_0_0::enable_if<(! ArduinoJson670_0_0::IsWriteableString<TPrint>::value), unsigned int>::type = unsigned int]'

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:95:35:   required from 'size_t ArduinoJson670_0_0::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson670_0_0::ObjectRef; TDestination = char* [256]; size_t = unsigned int]'

C:\Priv\Arduino.IDE\exercise\sketch_dec19a_JSON_test\sketch_dec19a_JSON_test.ino:59:34:   required from here

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp:146:13: error: request for member 'write' in '((ArduinoJson670_0_0::JsonWriter<char* [256]>*)this)->ArduinoJson670_0_0::JsonWriter<char* [256]>::_writer', which is of non-class type 'char* [256]'

     _length += _writer.write(static_cast<uint8_t>(c));

             ^

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp: In instantiation of 'void ArduinoJson670_0_0::JsonWriter<TWriter>::writeRaw(const char*, const char*) [with TWriter = char* [256]]':

[... cut ...] ze_t = unsigned int]'

C:\Priv\Arduino.IDE\exercise\sketch_dec19a_JSON_test\sketch_dec19a_JSON_test.ino:59:34:   required from here

C:\Priv\Arduino.IDE\libraries\ArduinoJson\src/ArduinoJson/Json/JsonWriter.hpp:138:13: error: request for member 'write' in '((ArduinoJson670_0_0::JsonWriter<char* [256]>*)this)->ArduinoJson670_0_0::JsonWriter<char* [256]>::_writer', which is of non-class type 'char* [256]'

     _length += _writer.write(reinterpret_cast<const uint8_t *>(begin),

             ^

```

bblanchon, 2018-12-22:
Hi @mlesn,

Are you sure you wrote:

```c++
char json_string[256];
serializeJson(root, json_string);
```

and not

```c++
char* json_string[256];
serializeJson(root, json_string);
```

?

Please check.

Regards,
Benoit

mlesn, 2018-12-26:
Hello,

> Please check.

Oh.. you are absolutely right - it was my mistake. 
Thank you!

Regards, M