bblanchon/ArduinoJson #874: InvalidInput after 5 successfull deserialization

Rexima, 2018-12-21:
Hello,

i want to send some data over the serial port with my own C# program.
But i get after 5 successfull transfers this error: InvalidInput

This is my Arduino Code:
`void loop(void) 
{ 
  while (!Serial) continue;
  
  if (millis() - runTime >= 500) { // Execute every 500ms
    runTime = millis();

    StaticJsonDocument<200> doc;
    DeserializationError error = deserializeJson(doc, Serial);  
    
    if (error) {
      Serial.print("Deserialize failed: ");
      Serial.println(error.c_str());
      return;
    }
    
    JsonObject root = doc.as<JsonObject>();`

And with my C# Client i send every 500ms a json string to my arduino.

bblanchon, 2018-12-22:
Hi @Rexima,

I see nothing wrong in the snippet above, and I'm pretty confident that it's not a bug in the parser.
Can you double check the C# program?

Depending on the transmission speed, it could also be a communication issue.
To check this hypothesis, you can read the input in a temporary buffer and print it if parsing fails.
Alternatively, you could use a logging proxy as suggested in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

Regards,
Benoit

gabrypulzio, 2018-12-22:
Hi @Rexima, make sure you aren't running out of memory. I had a similar issue and I found out I forgot to call clear() in the json buffer after parsing and using a JSON string.

closed 2019-01-07 by bblanchon