bblanchon/ArduinoJson #875: v6 API how to dumb JSONobject to Serial?

joeybab3, 2018-12-23:
I'm working on one of my libraries that utilizes the ArduinoJSON library and have been experimenting with the v6 library. As of right now, I'm getting `null` returned when attempting to access parts of the JSONObject.

I am most likely just using the wrong syntax, but I can't think of a way to check how the object is currently set up.

The JSON I'm parsing looks like this:

```
{
  "results": [
    {
      "id": "111111",
      "status": "ok",
      "label": "LABEL_HERE"
    }
  ]
}
```

I am attempting to access it like this(works in v5):

```
  DynamicJsonDocument jsonBuffer;
  DeserializationError error = deserializeJson(jsonBuffer, response);
  JsonObject root = jsonBuffer.as<JsonObject>();
  String id = root["results"][0]["id"];
  String label = root["results"][0]["label"];
  String status = root["results"][0]["status"];
  Serial.println("Id: "+id);
  Serial.println("Label: "+label);
  Serial.println("Status: "+status);
```

All 3 are returned as `null`.

bblanchon, 2018-12-23:
Hi @joeybab3,

The syntax is correct, as you can see in this demo:
https://wandbox.org/permlink/33lRrlVCoEHfey6f

However, you didn't check the value of [`DeserializationError`](https://arduinojson.org/v6/api/misc/deserializationerror/?utm_source=github&utm_medium=issues); the deserialization probably failed.

Regards,
Benoit

joeybab3, 2018-12-23:
The deserialization doesn't fail, I have the error code from the example as follows:
```
if (error) {
    Serial.print(F("Parsing JSON failed: "));
    Serial.println(error.c_str());
    return false;
  }
```
I was just being dumb and trying to omit things to make the issue shorter.

bblanchon, 2018-12-24:
Could you be out of memory?

What happens if you don't use `String`?

```c++
DynamicJsonDocument jsonBuffer;
DeserializationError error = deserializeJson(jsonBuffer, response);
JsonObject root = jsonBuffer.as<JsonObject>();
Serial.println(root["results"][0]["id"].as<char*>());
Serial.println(root["results"][0]["label"].as<char*>());
Serial.println(root["results"][0]["status"].as<char*>());
```

Regards,
Benoit

joeybab3, 2019-01-03:
Still nothing with the code you provided. And I don't think it's running out of memory because the sketch keeps running fine.

bblanchon, 2019-01-03:
What does the [`DeserializationError`](https://arduinojson.org/v6/api/misc/deserializationerror/?utm_source=github&utm_medium=issues) says?

joeybab3, 2019-01-03:
The error doesn’t print in the 
‘
If(error)
{
‘
Lines, is there a way I should be checking it’s value otherwise?

bblanchon, 2019-01-07:
@joeybab3, did you solve your problem? Can we close this issue?

joeybab3, 2019-01-10:
I've been trying to figure out what might be causing that, but even serial.printing the output and copying that directly to the wandbox demo you linked earlier parses correctly. 

Is it possible there's some sort of output that is not getting printed causing this error?

bblanchon, 2019-01-29:
Last time someone had a similar problem, it was because the input contained only space.
Could it be your case too?

joeybab3, 2019-01-31:
Only space meaning what? like just whitespace?

bblanchon, 2019-02-01:
Not only. It could also be a tab (`\t`), a carriage-return (`\r`), or a line-break (`\n`).
If your input only consists of these, deserializeJson() returns IncompleteInput.

joeybab3, 2019-02-01:
Oh I’m not sure that’s the case then, as I copied and pasted the output from serial moniter and just edited the values above.

bblanchon, 2019-02-02:
Hi @joeybab3,

The serial monitor sends a "\r\n" when you press enter.

That's exactly the problem we had in #862.
ArduinoJson stops reading the stream as soon as it sees the closing brace (`}`).
Next time you call `deserializeJson()`, the input only contains "\r\n", so it returns `IncompleteInput`.

The solution is to add a loop to flush the Serial buffer, as shown in the case-study "Recursive Analyzer" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues):

```c++
// Flushing Serial input buffer
while (Serial.available())
  Serial.read();

// Waiting for a JSON document in Serial
while (!Serial.available())
  delay(50);
```

Regards,
Benoit

joeybab3, 2019-02-06:
Sorry that was not what I meant by serial moniter, didn’t mean to confuse you.

I am using serial moniter to view the output of the variable “response” which is pulled off the internet. It has worked fine in the past with this exact input but now fails.

bblanchon, 2019-02-07:
@joeybab3, I really want to help you.
Can you provide an [MCVE](https://stackoverflow.com/help/mcve) that I compile an run myself?

joeybab3, 2019-02-07:
Yes, do you have an ESP32 handy? If so I'll post the code here and send you one of my authorization tokens privately so you are able to mirror my setup.

bblanchon, 2019-02-09:
I have an [Adafruit Feather HUZZAH ESP8266](https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/overview).
Can I reproduce your setup with that ?

closed 2019-02-26 by bblanchon