bblanchon/ArduinoJson #876: ESP32 WROVER External RAM Allocation Support

czuvich, 2018-12-26:
First of all, this library is awesome!  I am developing an Arduino ESP32 project using the WROVER module.  The WROVER chip (https://www.adafruit.com/product/3384) supports external RAM (up to 4MB), and I'd like to use  DynamicJsonBuffer to allocate memory to the external RAM module.  I'm making calls to a JSON backend that returns large(ish) JSON responses, and ideally, I would like to reserve internal RAM for ESP32 core functions. 

You can specify the memory strategy on the ESP32 for malloc calls to potentially use external RAM; however, the compiled ESP32 arduino core requires explicit allocation to external memory.  I've created a custom allocator that uses the `heap_caps_malloc(..., MALLOC_CAP_SPIRAM)` call; however, I don't think I can specify how to allocate the DynamicJsonBuffer internal allocations.

I know it's very platform specific, but 1) Is there a way to configure ArduinoJson to support ESP32 external allocation?  2) If not, it'd be great to have a way to do that :)

bblanchon, 2018-12-27:
Hi @czuvich,

Yes, in ArduinoJson v5, you can replace the allocator of `DynamicJsonBuffer`.

Here is the definition of `DynamicJsonBuffer`:

```c++
typedef Internals::DynamicJsonBufferBase<Internals::DefaultAllocator> DynamicJsonBuffer;
```

As you can see, it's a `typedef` of a template class named `DynamicJsonBufferBase` that take an allocator class as a parameter. Here is the definition of `DefaultAllocator`:

```c++
class DefaultAllocator {
 public:
  void* allocate(size_t size) {
    return malloc(size);
  }
  void deallocate(void* pointer) {
    free(pointer);
  }
};
```

To use the external memory, we need to substitute this class with one that uses the right functions. If I understand correctly, it should be something like that:

```c++
class SpiRamAllocator {
 public:
  void* allocate(size_t size) {
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
  }
  void deallocate(void* pointer) {
    heap_caps_free(pointer);
  }
};

typedef ArduinoJson::Internals::DynamicJsonBufferBase<SpiRamAllocator> SpiRamJsonBuffer;
```

I'd be very interested in knowing if it works, so **please keep me informed**.

It's currently not possible to do something like that with ArduinoJson version 6. However, if you confirm that it's useful, I can add something similar.

Thank you very much for this excellent question.

Regards,
Benoit

czuvich, 2018-12-28:
@bblanchon The solution worked flawlessly!  Thank you for the detailed response!  I have some fallback code in the allocator just in case SPI ram allocation failed (in case you're interested), and you can just call `free(p)` instead of the explicit `heap_caps_free(p)` in IDF.

```
void* ret = heap_caps_malloc(sz, MALLOC_CAP_SPIRAM);
  if (ret)
    return ret;
  else
    return malloc(sz);
```
It would be great to see this capability in ArduinoJson v6.  I think it'll be a huge help as more developers start using ESP32.  Again, thank you so much!

bblanchon, 2018-12-30:
Hi @czuvich,

Thank you very much for the feedback.
I'll add something similar in version 6.

> If we have the capability to leverage external memory on the ESP32 with ArduinoJson, I think that would be a huge help.

I'm puzzled by this sentence. Do you expect more from me than adding the customizable allocator in v6?

Regards,
Benoit

czuvich, 2018-12-30:
The allocator is perfect.

bblanchon, 2018-12-31:
Got it!
I'll leave this issue opened as a reminder.

sticilface, 2019-03-17:
oh my... i searched for custom allocator and didn't find it.  oops...

bblanchon, 2019-03-17:
I just added `BasicJsonDocument` in the 6.x branch.
You can use it like that:

```c++
struct SpiRamAllocator {
  void* allocate(size_t size) {
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
  }
  void deallocate(void* pointer) {
    heap_caps_free(pointer);
  }
};

using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
```

It will be available in 6.10.

closed 2019-03-17 by bblanchon