bblanchon/ArduinoJson #880: [v6.7.0-beta] DynamicJsonDocument not expanding

moose4lord, 2019-01-01:
I have been kicking the tires on the v6 beta release and noticed that the DynamicJsonDocument document isn't expanding from its initial size. Serialization simply fails with a "noMemory" error even when there's plenty of room on the heap. After poking around the documentation, I did see this mentioned:

> In older versions, DynamicJsonDocument was able to grow if needed. Starting with version 6.7.0, DynamicJsonDocument has a fixed capacity, just like StaticJsonDocument.

Are you abandoning the dynamic JsonDocument, or is this a temporary setback?

bblanchon, 2019-01-02:
Hi @moose4lord,

I deliberately made this change because it was causing more harm than good:

* it wasted a lot of heap memory
* it caused heap fragmentation
* it was in complete contradiction with the fixed-allocation / no-fragmentation philosophy of the library
* it forced me to keep virtual functions
* it pretended to help the users but was really a ticking bomb

Unless there is a very good reason to go back to the old behavior, I intend to keep it that way.
Do you have a use-case that *absolutely* requires the automatic expansion?

However, I'm not sure it was wise to keep the default constructor; maybe the argument should be mandatory. What do you think?

Regards,
Benoit

moose4lord, 2019-01-02:
I can't say I have a use case that _absolutely_ needs the dynamic expansion, but it certainly was convenient. My app uses one JSON doc, but it can have a wide range of sizes, from a few dozen bytes to a few thousand bytes. Performance wasn't a concern for me, so i just let DynamicJsonDocument handle it. I guess I'll have to put more thought into calculating a static size  that will work for me.

If it were up to me, I think I'd get rid of the DynamicJsonDocument version of JsonDocument altogether. It looks too much like v5's DynamicJsonBuffer, and would surely confuse user's migrating from v5 to v6 (like it did me ;).

bblanchon, 2019-01-30:
In [version 6.8.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.8.0-beta), I removed the default constructor.
You now have to explicitly set the capacity of `DynamicJsonDocument`, which should be less confusing.

closed 2019-01-30 by bblanchon