bblanchon/ArduinoJson #882: Create Nested array in loop for OpenHAB

Tobi130, 2019-01-03:
Hello,

I'm trying a project with ESP32 and IR sensor MLX90640 which output is an array with size 768. I want to send this array in JSON format via MQTT to a OpenHAB server. OpenHAB needs the information in the following format: {"Temperature":{v1:23.4,v2:24.8,v3:24.1,v4:24.9,v5:23.7,.........}} up to v768.
I didn't get how to create the nested arrays v1, v2, ... ,v768 in a for loop in the array Temperature and also if it's possible to write after "Temperature" a { instead of [. OpenHAB isn't able to work with [. I'm working with ArduinoJson 5.13.4.

This is my actual code. Output isn't structured and looks like {"Temperature":[24.7,23.7,25.3,25.3,25.4,25.5,........}

    const size_t bufferSize = JSON_ARRAY_SIZE(768) + JSON_OBJECT_SIZE(1);
    DynamicJsonBuffer jsonBuffer(bufferSize);
    JsonObject& JSONencoder = jsonBuffer.createObject();

    JsonArray& Temperature = JSONencoder.createNestedArray("Temperature");

  for (int x = 0 ; x < 768 ; x++)
  {
    float temp;
    temp = int((mlx90640To[x] + 0.5)*10);
    mlx90640To[x] = temp / 10;
    Temperature.add(mlx90640To[x]);
  }

    char JSONmessageBuffer[5000];
    JSONencoder.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
    Serial.println("Sending message to MQTT topic..");

    if (client.publish("IR_MQTT", JSONmessageBuffer) == true) {
      Serial.println("Success sending message");
    } else {
      Serial.println("Error sending message");
    }
Does anybody have ideas?

bblanchon, 2019-01-04:
Hi @Tobi130,

Here is what our dear [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/) says:

```c++
const size_t bufferSize = JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(5);
DynamicJsonBuffer jsonBuffer(bufferSize);

JsonObject& root = jsonBuffer.createObject();

JsonObject& Temperature = root.createNestedObject("Temperature");
Temperature["v1"] = 23.4;
Temperature["v2"] = 24.8;
Temperature["v3"] = 24.1;
Temperature["v4"] = 24.9;
Temperature["v5"] = 23.7;
```

I clearly shows how to create the nested object your looking for.
So, I guess your question is "How to format the key?".

Here is one solution using `sprintf()`:

```c++
for (int x = 0 ; x < 768 ; x++) {
  char key[8];
  sprintf(key, "v%d", x+1);
  Temperature[key] = mlx90640To[x];
}
```

Regards,
Benoit

Tobi130, 2019-01-04:
Hi @bblanchon,

thank you for you explanation. I tried with the assistant, but forgot the "". So it aborted. 
Now I also understand the difference between Object and Array. 
With the new code I need a JSONmessageBuffer over 9000 and get a reset on the esp32, but that's another story.

Thank you!

closed 2019-01-04 by Tobi130