bblanchon/ArduinoJson #883: unable to create and use two JsonBuffer

simonfongnt, 2019-01-06:
Platform: ESP-12E
Compiler: Arduino 1.8.8 with ArduinoJson 5.13.3

Hi, i am trying to create two jsonbuffer and use them by defining a class.
```
class JsonParams{
    private:
        JsonObject *jsonObjectPtr;
    public:
        void createjObject(String jString){
            StaticJsonBuffer<JSON_ARRAY_SIZE(400)> jBuffer;
            JsonObject& jObject = jBuffer.parseObject(jString);
            jsonObjectPtr = &jObject;
        }
        JsonObject& jsonObject() {
            return *jsonObjectPtr;
        }
        String jsonString() {
            String jString;
            (*jsonObjectPtr).printTo(jString);
            return jString;
        }
};
```
The reason to create the class is the jsonobject can access and edit the parameters in matrix form without further declaration of parameters:
```
  StaticJsonBuffer<JSON_ARRAY_SIZE(400)> jBuffer;
  JsonObject& jObject = jBuffer.parseObject(jString);
  Serial.println(jObject["device"]["setting1"]);
  jObject["device"]["setting1"] = 123
```
or with the class,
```
  jsonParams.createjObject(jsonstring);
  Serial.println(jsonParams.jsonObject()["device"]["setting1"]);
  jsonParams.jsonObject()["device"]["setting1"]  = 123;
```
Although this was an idea, I understand JsonBuffer do not last forever. I personally wish it could lasts for a mainloop so that the jsonobject can be updated with string at every beginning of the loop. If it is impossible, at least the class can print out the updated String for global usage.

However, I read a [suggestion](https://arduinojson.org/v5/faq/how-to-reuse-a-jsonbuffer/) and hope the class can create multiple jsonbuffer for my purpose but I had an issue:

Suppose:
```
statusjstring = {"on":0,"ir":0,"sonic":0,"pir":0}
paramsjstring = {smart":{"on":0,"ir":{"Von":512,"Voff":512,"sampling":10,"delay":300}}}
```
Code:
```
    JsonParams statusjParams;
    JsonParams paramsjParams;

    statusjParams.createjObject(statusjstring);
    paramsjParams.createjObject(paramsjstring);

    statusjParams.jsonObject().printTo(Serial);
    Serial.println(); 
    paramsjParams.jsonObject().printTo(Serial);
    Serial.println(); 
```
Output:
```
{smart":{"on":0,"ir":{"Von":512,"Voff":512,"sampling":10,"delay":300}}}
{smart":{"on":0,"ir":{"Von":512,"Voff":512,"sampling":10,"delay":300}}}
```
Seem the jsonbuffer is overwritten the previous even JsonBuffer is defined in class. I work around the size of StaticJsonBuffer up to 4000 but performed the same.

The reason might be the library was supposed to do it or  I do not familiar with arduino or C++. At very least, defining parameters for every item in json is my last option.

I wonder if you have any idea to this problem. Inside the device, it is possible for me to group all json into one. But with communication, I still wish to compare incoming json with device json data piece by piece which requires two JsonBuffer exists simultaneously.

Please notice me if you want to have further check or info from my side.

simonfongnt, 2019-01-06:
I think I should 1st give a try on ArduinoJson v6.7

bblanchon, 2019-01-07:
Hi @simonfongnt,

The `JsonBuffer` should be a member of the class, not a local variable in `createjObject()`, otherwise `jsonObjectPtr` points to a destructed object. You can see an example in the case study "JSON-RPC with Kodi" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

You're right, you should try ArduinoJson 6 because it's easier to use a [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/?utm_source=github&utm_medium=issues) than a [`JsonBuffer`](https://arduinojson.org/v5/api/jsonbuffer/?utm_source=github&utm_medium=issues) as a class member.

Regards,
Benoit

simonfongnt, 2019-01-09:
Thanks for the great help. 
I will design the project with v6.

closed 2019-01-09 by simonfongnt