bblanchon/ArduinoJson #884: How to convert string into array for printing Json Object Values

Nesans, 2019-01-07:
`{
    "id": 670574,
    "name": "sample",
    "description": null,
    "latitude": "0.0",
    "longitude": "0.0",
    "created_at": "2019-01-07T12:19:15Z",
    "elevation": null,
    "last_entry_id": null,
    "public_flag": true,
    "url": null,
    "ranking": 30,
    "metadata": null,
    "license_id": 0,
    "github_url": null,
    "tags": [],
    "api_keys": [
        {
            "api_key": "3XEYCUZPDRRR0E8Y",
            "write_flag": true
        },
        {
            "api_key": "RXFRRG9T1USCYC59",
            "write_flag": false
        }
    ]
}`

The above given JSON document has to be parsed. Since we are using nodemcu and httpclient library, we are getting the reply from api as string. We are able to print id into the serial monitor, but we don't know how to print two api_key which is nested into the api_keys object. Please find our code below. Kindly help.

```
HTTPClient http; //Object of class HTTPClient
    http.begin("http://api.thingspeak.com/channels.json? api_key=V3KVIFDDALMZ6TDQ&name=Naveen&public_flag=true&field1=Dipika");
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");
    int httpCode = http.POST("api_key=V3KVIFDDALMZ6TDQ");
    Serial.println(httpCode);
    String payload = http.getString();
    Serial.println(payload);
    http.end(); //Close connection
    delay(1000);
    DynamicJsonBuffer jsonBuffer;
    JsonObject& channel = jsonBuffer.parseObject(payload);
    int ID = channel[String("id")];
    const char* read = channel[String("api_keys")][0];
    Serial.println(ID);
    Serial.println(read);

    // Test if parsing succeeds.
    if (!channel.success()) {
    Serial.println("parseObject() failed");
    return;
```

Nesans, 2019-01-08:
Thank you so much benoit. It works ! :)

bblanchon, 2019-01-09:
You're welcome @Nesans.
Thanks for using ArduinoJson.
Don't forget to add a [star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the library ;-)

closed 2019-01-09 by bblanchon