bblanchon/ArduinoJson #885: Unable to copy a JsonOjbect to old jsonobject in v6

simonfongnt, 2019-01-08:
Platform: ESP-12E
Compiler: Arduino 1.8.8 with ArduinoJson 6.7.0

Hi, i am digging around the beta version. Things are well-defined until I have this issue. I wish to create a function to update existing json with incoming json as follows:
```
JsonObject updatejson(JsonObject oldjobject, JsonObject newjobject){  
    for (auto kv : newjobject) {
        const char* key   = kv.key().c_str();
        int len = JsonObject(newjobject[key]).size();
        if (len > 0){
            Serial.println(F("len > 0, "));
            JsonObject jobject          = newjobject[key];
            JsonArray jarray            = newjobject[key];
            if (!jobject.isNull()){
                JsonObject oldnode      = oldjobject[key];
                JsonObject newnode      = newjobject[key];
                Serial.println(key);
//                oldjobject[key] = updatejson(oldnode, newnode);        
                JsonObject jupdate = updatejson(oldnode, newnode);  // 4th trial: define new jobject to hold the result - recurrsive function works: jupdate has jsonobject
                oldjobject[key] = jupdate;                          // <-- cannot copy jsonobject to oldobject?
            }
            else if (!jarray.isNull()){
                JsonArray oldarr        = oldjobject[key];
                JsonArray newarr        = newjobject[key];
                for (auto value : newarr) {
                    JsonObject oldnode  = jarray[value];
                    JsonObject newnode  = jarray[value];
//                    oldjobject[key][value] = updatejson(oldnode, newnode);
                    JsonObject jupdate = updatejson(oldnode, newnode);
                    oldjobject[key] = jupdate;
                }    
            }else{
                Serial.println(F("Both object & array are null"));
                // no matching node - no update  
            }
        }
        else{
            Serial.println();
            if (oldjobject.containsKey(key)){
                JsonVariant oldjvar = oldjobject[key];
                JsonVariant newjvar = newjobject[key];
                if (oldjvar.is<int>() and newjvar.is<int>()){
//                    oldjobject[key] = kv.value().as<int>();    // 1st trial: use the value on new json  to update
//                    oldjobject[key] = 123;                     // 2nd trial: use integer to update
//                    oldjobject[key] = "123";                   // 3rd trial: use string to update
                    serializeJson(oldjobject[key], Serial);
                }else if (oldjvar.is<const char*>() and newjvar.is<const char*>()){
                    oldjobject[key] = kv.value().as<char*>();
                    serializeJson(oldjobject[key], Serial);
                }else{
                    Serial.print(F("Inconsistant incoming value..."));
                }
            }else{              
                Serial.print(F("key does not exists..."));
            }
        }
    }
    return oldjobject;
}
```
JSONs:
```
oldjson: -> {"type":"LIGHTING","name":"The kitchen light","test":{"host":"http://webserver.com/domain","port":"80","id":"LoginID","pw":"LoginPW"},"ota":{"host":"http://webserver.com/domain","port":80,"id":"LoginID","pw":"LoginPW"}}

newjson -> {"type":"LIGHTING","name":"The kitchen light","test":{"host":"http://webserver.com/domain","port":"81","id":"ABC","pw":"12345678"},"ota":{"host":"http://webserver.com/domain","port":81,"id":"ABC","pw":"12345678"}}
```
Output:
`{"type":"LIGHTING","name":"The kitchen light","test":{},"ota":{}}`

I think my issue is cannot copy the jsonobeject from the function to existing/old jsonobject.
```
                    JsonObject jupdate = updatejson(oldnode, newnode);
                    oldjobject[key] = jupdate;
```
this might be caused by fixed memory of json document but i am not sure.

I did try to use reference of jsonobject without success
`JsonObject *oldjpointer = &oldjobject,`
i could use the jsonobject reference to update the json but ran into trouble exporting it into string in v5.

I have provided a test code as well:
[arduinojson.zip](https://github.com/bblanchon/ArduinoJson/files/2735463/arduinojson.zip)

Maybe there is a simple explaination or solution for my case. Please let me know if you want any further info from me.

simonfongnt, 2019-01-08:
Instead of 
```
                    JsonObject jupdate = updatejson(oldnode, newnode);
                    oldjobject[key] = jupdate;
```
I tried the following code which could fix the issue:
```
                String tempjstring;
                DynamicJsonDocument updatejdoc;
                JsonObject updatejobject = updatejdoc.to<JsonObject>(); 
                updatejobject = updatejson(oldnode, newnode);
                serializeJson(updatejobject, tempjstring); 
                DynamicJsonDocument tempjdoc;
                deserializeJson(tempjdoc, tempjstring);
                oldjobject[key] = tempjdoc.as<JsonObject>();  
```
but I am worry this would incur memory issue in a long run. 
Sorry to tell that I declared a global DynamicJsonDocument for oldjdoc and oldjobject for rest of my code.
I suppose the oldjobject would hold forever with this setting.

bblanchon, 2019-01-09:
Hi @simonfongnt,

That's a lot of code!
I tried to follow, but I didn't understand what you want to accomplish with `updatejson()`.
Please tell me what you want to do and I'll tell you how to do it.

Meanwhile, here is what you need to know when it comes to copying.

[JsonDocument](https://arduinojson.org/v6/api/jsondocument/?utm_source=github&utm_medium=issues) has value semantics. If you assign a document to another, you make a copy:

```c++
DynamicJsonDocument doc2 = doc1; // copy the whole document
```

[JsonObject](https://arduinojson.org/v6/api/jsonobject/?utm_source=github&utm_medium=issues), however, has reference semantics. If you assign a JsonObject to another, you just change the reference.

```c++
JsonObject obj2 = obj1; // copy only the reference
// obj2 and obj2 point to the same structure in memory
```

If you want to duplicate the object, you must call [JsonObject::copyFrom()](https://arduinojson.org/v6/api/jsonobject/copyfrom/?utm_source=github&utm_medium=issues):

```c++
DynamicJsonDocument doc2(1024);
JsonObject obj2 = doc2.to<JsonObject>();
obj2.copyFrom(obj1);  // copy the object
```

`JsonObject::copyFrom()` was missing from the documentation, I just added it.

Regards,
Benoit

simonfongnt, 2019-01-09:
Sorry to write that tons of code without proper comments....
The code was aimed to keep built-in json in consistent format while incoming json can edit its value only.
For example, built-in Json:
```
{
  "type": "Device Type",
  "name": "Device Name",
  "ota": {
    "host": "http://webserver.com/domain",
    "port": 80,
    "id": "LoginID",
    "pw": "LoginPW"
  }
}
```
Incoming Json can be:
```
{
  "name": "The kitchen light",
  "ota": {
    "port": 81,
  }
}
```
New jObject cannot be added and value type has to be matched.

Seem I have something to achieve this but I am not sure if there is some potential issue. I have attached the example code [arduinojsonV1.zip](https://github.com/bblanchon/ArduinoJson/files/2741547/arduinojsonV1.zip)
 as well. You may check it if you really have time.

One thing I noticed the JsonArray loop method is different from v5, in which the "value" becomes the JsonVariant instead of index of JsonArray. I tried to create an "int index" before the loop in order to access the array at specific position. Maybe you'd hv some idea of getting the index of array like v5.

Thanks for this awesome project. Save a tons of time!

simonfongnt, 2019-02-10:
Hi @bblanchon ,
Sorry to reply it late. Seem to me the DynamicJsonDocument requires specific size in v6.8.
Thanks for the neat suggestion for my issue. I think the following was a way for updateArray but might not be what you're expecting.

void updateObject(JsonObject oldObject, JsonObject newObject) {
    for (JsonPair kv : newObject) 
        updateVariant(oldObject[kv.key().c_str()], kv.value());
}

void updateArray(JsonArray oldArray, JsonArray newArray) {
    int index = 0;                                                        
    for (auto value : newArray) {                                           
        updateVariant(oldArray[index], newArray[index]);
        index++;                                                            
    }
}

void updateVariant(JsonVariant oldVariant, JsonVariant newVariant) {
    if (newVariant.is<JsonObject>())
        updateObject(oldVariant, newVariant);
    else if (newVariant.is<JsonArray>())
        updateArray(oldVariant, newVariant);
    else if ((oldVariant.is<int>() and newVariant.is<int>())
        ||(oldVariant.is<const char*>() and newVariant.is<const char*>())){
        oldVariant.set(newVariant);
    }
}t index = 0;                                                          
    for (auto value : newArray) {                                           
        updateVariant(oldArray[index], newArray[index]);
        index++;                                                            
    }
}

bblanchon, 2019-02-11:
The `updateArray()` above will only work if the two arrays have the same size.
It's not easy to decide what to do with arrays... should you replace or append the element?
That's up to you to decide, that's why I left that as an "exercise to the reader".

Do you need further assistance? If not, we should close this issue.

closed 2019-02-11 by simonfongnt