bblanchon/ArduinoJson #888: Compilation failure in ESP32 with StringTraits::Reader not found

FLamparski, 2019-01-12:
I get this error every time I try to compile my project using this library. The use-case is decoding JSON messages coming through BLE. I am on Windows 10.

Target platform: ESP32
Compiler: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0

Example code:
```c++
#include <Arduino.h>
#include <ArduinoJson.h>

StaticJsonBuffer<64> JSONBuffer;

void setup() {
    std::string json = "{\"ts\":12345}";

    JsonObject& msg = JSONBuffer.parseObject(json);

    long ts = msg["ts"];
}

void loop() {
}
```

Compiler output:

```
In file included from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7:0,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from c:\Users\ftwie\Documents\Projects\MobilityGATTDemo\BLE_uart.ino:2:

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp: In instantiation of 'struct ArduinoJson::Internals::JsonParserBuilder<ArduinoJson::Internals::StaticJsonBufferBase, std::__cxx11::basic_string<char>, void>':

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66:   required by substitution of 'template<class TJsonBuffer, class TString> typename ArduinoJson::Internals::JsonParserBuilder<TJsonBuffer, TString>::TParser ArduinoJson::Internals::makeParser(TJsonBuffer*, TString&, uint8_t) [with TJsonBuffer = ArduinoJson::Internals::StaticJsonBufferBase; TString = std::__cxx11::basic_string<char>]'

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:86:33:   required from 'ArduinoJson::JsonObject& ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(TString&, uint8_t) [with TString = std::__cxx11::basic_string<char>; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; uint8_t = unsigned char]'

c:\Users\ftwie\Documents\Projects\MobilityGATTDemo\BLE_uart.ino:9:50:   required from here

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:73:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<std::__cxx11::basic_string<char>, void>'

   typedef typename StringTraits<TString>::Reader InputReader;

                                                  ^

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:74:50: error: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<std::__cxx11::basic_string<char>, void>'

   typedef JsonParser<InputReader, TJsonBuffer &> TParser;

                                                  ^

In file included from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7:0,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from c:\Users\ftwie\Documents\Projects\MobilityGATTDemo\BLE_uart.ino:2:

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp: In instantiation of 'ArduinoJson::JsonObject& ArduinoJson::Internals::JsonBufferBase<TDerived>::parseObject(TString&, uint8_t) [with TString = std::__cxx11::basic_string<char>; TDerived = ArduinoJson::Internals::StaticJsonBufferBase; uint8_t = unsigned char]':

c:\Users\ftwie\Documents\Projects\MobilityGATTDemo\BLE_uart.ino:9:50:   required from here

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:86:33: error: no matching function for call to 'makeParser(ArduinoJson::Internals::StaticJsonBufferBase*, std::__cxx11::basic_string<char>&, uint8_t&)'

     return Internals::makeParser(that(), json, nestingLimit).parseObject();

                                 ^

In file included from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7:0,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from c:\Users\ftwie\Documents\Projects\MobilityGATTDemo\BLE_uart.ino:2:

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66: note: candidate: template<class TJsonBuffer, class TString> typename ArduinoJson::Internals::JsonParserBuilder<TJsonBuffer, TString>::TParser ArduinoJson::Internals::makeParser(TJsonBuffer*, TString&, uint8_t)

 inline typename JsonParserBuilder<TJsonBuffer, TString>::TParser makeParser(

                                                                  ^

C:\Users\ftwie\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:96:66: note:   substitution of deduced template arguments resulted in errors seen above

exit status 1
```

FLamparski, 2019-01-12:
The code compiles if I first get the `c_str()` of the incoming `std::string`. I'm leaving this issue open, though, as `std::string`s should be accepted IMO.

bblanchon, 2019-01-12:
Hi @FLamparski,

ArduinoJson supports `std::string` but it automatically disables it when `ARDUINO` is defined.
However, you can define [ARDUINOJSON_ENABLE_STD_STRING](https://arduinojson.org/v5/api/config/enable_std_string/?utm_source=github&utm_medium=issues) to `1` to force the support for `std::string`.

Regards,
Benoit

FLamparski, 2019-01-17:
That seems to work. Thanks.

bblanchon, 2019-01-19:
You're welcome @FLamparski.
Thank you for using ArduinoJson.

closed 2019-01-19 by bblanchon