bblanchon/ArduinoJson #889: Sudden issues when trying to print Json Array to Serial (possibly related to a struct used elseware)

kruegersc4, 2019-01-12:
Hello,

I have a program on an Arduino Uno using arduinoJson (5.13.14), a DHT sensor with the DHT.h library, and an LCD screen using the LiquidCrystal.h library. The program works perfectly fine with all other lines of code, and when parsing my array it reads successful with Success(). But when I try to print the Json array 'Environment' with Environment.printTo(Serial); , or try to print one of the objects within, everything is suddenly inconsistent. My program suddenly fails to read from my DHT sensor and the LCD screen stops displaying. The serial may intermittently print the DHT read failure warning or print the Environment array with zeros in the objects, or pieces of these.

I tried to increase the buffer size, which I think is already very large for this since I was using it to deserialize a much bigger array, but it didn't help. I also tried to decrease its size. I also tried printing to a character array and then printing to the serial. If I take out the Environment.printTo line and add print lines with characters or other variables, the program works fine.

Here's the relevant code:

```
//newdata, CurrentTemp (float), CurrentHumidity (integer), and bufferSize, declared elseware

void SendData() {
  if (newdata) { // only happens if temperature/humidity changes
    StaticJsonBuffer<bufferSize> jsonBuffer;
    
    JsonArray& Environment = jsonBuffer.createArray();
    JsonObject& temp = Environment.createNestedObject();  
    JsonObject& humi = Environment.createNestedObject();
  
    temp["0"] = CurrentTemp;
    humi["0"] = CurrentHumidity;

    if (Environment.success()) {  
      Environment.printTo(Serial);
    }
  }
}
```

Also, the printing above works fine if I take out the line of code noted in the following deserialization function, where I deserialize an unrelated JSON string to a global struct 'OutputDatas'. But the struct works fine if I take out the printing line - so either one works if the other is not there:

```
  if (Serial.available() > 0) {
  
  StaticJsonBuffer<bufferSize> jsonBuffer;
  
  JsonArray& cReadArray = jsonBuffer.parseArray(Serial);
  
  if (cReadArray.success()) {
    int i = 0;
    for (JsonVariant& elem : cReadArray) {
      if (elem.is<JsonObject>())  { //will screen the objects more in the future

        const char* ontime = elem["ontime"];
        float ontimef = HourMinConversion(ontime); //converts time in 00:00:00 style to float
        
        const char* offtime = elem["offtime"];
        float offtimef = HourMinConversion(offtime);

        const char* type = elem["type"]; 

        OutputDatas[i] = { type, ontimef,offtimef, elem["target"], elem["differential"]}; //**Printing Works if I take out this line**

      }
     i++;
    }
  
  } else {
  // Serial.println("Failed to parse incoming JSON data");
  }
  
  }
```

Here is the code I use before this to create the struct as a global variable:

```
typedef struct {
  const char* type; 
  float ontime;
  float offtime;
  int target;
  int differential;
} OutputData;

OutputData OutputDatas[6];
```

I'm starting to suspect that I'm messing up with the struct, or with multiple things... 

Thank you! Forgive me if this is silly, I'm a novice.

bblanchon, 2019-01-13:
Hi @kruegersc4,

I could have stopped reading at "everything is suddenly inconsistent." 😄 

When things don't make sense, it is usually a sign of a stack overflow.
The stack is the place where the program stores local variables and return addresses.
When the stack gets corrupted, the microcontroller behaves weirdly because it doesn't execute the program anymore but whatever is present in memory.

An Arduino UNO only has 2KB of RAM, you must use it very carefully.
In your case, you must make sure that the `JsonBuffer`, and all the other variables (not to mention the call stack) fit in the RAM.

Regards,
Benoit

kruegersc4, 2019-01-13:
Thanks for the good explanation. So it sounds like my first mistake was using a large placeholder buffer for a small array. Even though It gets deleted once it leaves scope, the struct, and all other global variables, are always in the memory once created, so together they probably cause this? I'll be doing some streamlining...

Another thing I'm struggling to understand, particularly with this library, are pointers. Is it okay that I have a const char pointer in my struct  ('type'), or will that leave remnants of the JSON objects in memory? I used a pointer because I noticed you did on a struct example, and it doesn't save the strings otherwise, but I realize I don't really know why I used it! (maybe I should just have different corresponding integers for different 'types' to make it simpler...)

Once again, thank you very much for your time.

bblanchon, 2019-01-14:
Hi @kruegersc4,

In this configuration, the pointers that ArduinoJson returns point to the `JsonBuffer`.
So, it is OK to store pointers in the struct, as long as you can guarantee that the `JsonBuffer` remains in memory.

If you want to store a copy of the string in the struct, use a `char[]` instead:

```c++
typedef struct {
  char type[16]; 
  float ontime;
  float offtime;
  int target;
  int differential;
} OutputData;
```

Then make a copy with `strcpy()`:

```c++
strlcpy(OutputDatas[i].type, elem["type"] | "N/A", 16);
```

You can see a complete example in [JsonConfigFile.ino](https://arduinojson.org/v5/example/config/?utm_source=github&utm_medium=issues).

If you are lost with pointers, references, stack, heap... you might be interested in reading "The missing C++ course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Regards,
Benoit

kruegersc4, 2019-01-16:
Great, thanks! I'll definitely consider the book. 

I realized I didn't need the char array in there and ended up replacing the char type; with an integer, with different numbers for different types, for simplicity's sake. So it looks like this:

```
typedef struct {
  int type;
  float ontime;
  float offtime;
  int target;
  int differential;
} OutputData;

OutputData OutputDatas[6];
```
Now, I know this next question is only tangentially related to ArduinoJSON... I'm still figuring out the stack overflow issues, and I noticed it also overflows if the Serial.println() is added to the function below, printing a simple string. Are print commands particularly memory hungry? Or is this probably just a build up of a lot of things in memory? 

```
for (int p = 0; p < sizeof(OutputDatas)/sizeof(OutputDatas[0]); p++) {
      if (OutputDatas[p].type == 3) { // If type is Humidifier
        BasicUpControl(CurrentHumidity, OutputDatas[p].target, OutputDatas[p].differential, p); //basic control function
        Serial.println("Controlled!"); //causes stack overflow
      }
      if (OutputDatas[p].type == 0) { // If type is Light
        Timer(true, currenthourmin, OutputDatas[p].ontime, OutputDatas[p].offtime, p);
      }

}
```

sizeof(OutputDatas) only gives 84 bytes, and BasicUpControl() and Timer() are pretty simple functions. I'm definitely keeping my ArduinoJSON buffers in local scope so I don't think it's that, but I thought I'd ask.

bblanchon, 2019-01-16:
Hi @kruegersc4,

I don't think `Serial.println()` uses a lot of memory.
The problem is the accumulation of all the variables in the program.
BTW, the FAQ contains [several tips to reduce the memory usage](https://arduinojson.org/v5/faq/how-to-reduce-memory-usage/?utm_source=github&utm_medium=issues).

Regards,
Benoit

kruegersc4, 2019-01-20:
I finally figured out the original issue and got the Environment.printTo(Serial) line to work. For the LCD screen, I had a lot of lines of code for lcd.print(). I changed the lines that were just unchanging strings to lcd.print(F())), and that apparently freed up enough memory. It took a little more program memory, but the LCD screen should be on a separate microcontroller eventually anyway.

Thanks for all of your help, Benoit.