bblanchon/ArduinoJson #891: serializing multiple structs within an array to JSON

schmonet, 2019-01-17:
Hi Benoit,

i have a similar question, so will add it to this thread.
I need to **serialize and deserialize some structs** conataining various elements, which **should be added by the user** in the Web-UI and sent back to the ESP8266 to be stored in the persistantly eeprom as struct or on the SPIFFS as JSON. I bought your eBook. Good work!

I have these struct and array to provide a **preset of color and control-settings** for my **lamp**-project, which is supposed to **behave** (soft on, soft of, motion detector) **differently, depending on the time of day** (NTP):

```c
typedef struct {
	uint32_t time; //time to next dimming point
	uint16_t h, s, v, w, ww; //lamp color, sat, hue and extra white + warm white
} dim_step_t;
```

```c
/*
dim_table 'time table' to specify dim_table
*/
typedef struct
{
  uint32_t clock; //time of day
  uint16_t stdby, dist, ldr; //standby time, motion detection distance, ambient light level threshold
  dim_step_t dim_step[4]; //as a preset only 4 dimming points will be populated, but the user should be able to add more or delete some
} dim_table_t;

dim_table_t dim_table[] = {
   //clock, stdby, dist, ldr, time, h, s, v, w, ww, time, h, s, v, w, ww, time, h, s, v, w, ww...
	{ 480, 20,  0,  50, {{0, 0, 0, 0, 0, 0},{3, 0, 0, 500, 0, 0},{30, 0, 0,  50, 0, 0},{ 30, 0, 0, 0, 0, 0}}},
	{ 720,  0,  1, 100, {{0, 0, 0, 0, 0, 0},{1, 0, 0, 100, 0, 0},{20, 0, 0, 100, 0, 0},{ 10, 0, 0, 0, 0, 0}}},
	{1080,  0,  1,  50, {{0, 0, 0, 0, 0, 0},{3, 0, 0,  30, 0, 0},{10, 0, 0,  30, 0, 0},{300, 0, 0, 0, 0, 0}}},
	{1260,  1, 20,   0, {{0, 0, 0, 0, 0, 0},{6, 0, 0,  10, 0, 0},{60, 0, 0,  10, 0, 0},{ 10, 0, 0, 0, 0, 0}}},
}; //clock = 480 defines the first timeslot in minutes of day from midnight in relation to NTP wall clock time
```

**this should be serialized with ArduinoJson to:**

```json
[
  { 
    "clock": 480, 
    "stdby": 20,
    "dist": 0, 
    "ldr": 50,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 500, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 50, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 30, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 720, 
    "stdby": 0,
    "dist": 1, 
    "ldr": 100,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 1, 
        "h": 0, 
        "s": 0, 
        "v": 100, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 20, 
        "h": 0, 
        "s": 0, 
        "v": 100, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 1080, 
    "stdby": 0,
    "dist": 1, 
    "ldr": 50,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 3, 
        "h": 0, 
        "s": 0, 
        "v": 30, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 30, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 300, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  },
  { 
    "clock": 1260, 
    "stdby": 1,
    "dist": 20, 
    "ldr": 0,
    "step": [
      { 
        "time": 0, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 6, 
        "h": 0, 
        "s": 0, 
        "v": 10, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 60, 
        "h": 0, 
        "s": 0, 
        "v": 10, 
        "w": 0, 
        "ww": 0 
      },
      { 
        "time": 10, 
        "h": 0, 
        "s": 0, 
        "v": 0, 
        "w": 0, 
        "ww": 0 
      }        
    ]
  }
]
```

I used https://arduinojson.org/v5/assistant/ to calculate the Buffersize for ESP8266 to 2916:

```c
const size_t bufferSize = 5*JSON_ARRAY_SIZE(4) + 4*JSON_OBJECT_SIZE(5) + 16*JSON_OBJECT_SIZE(6) + 620;
```

i wanted to implement the serialization of dim_table[] as follows:

```c
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonArray& root = jsonBuffer.createArray();
String json;

for (uint8_t i = 0; i < (4 - 1); i++) { 
	JsonObject& block = root.createNestedObject("block");
	block["clock"] = dim_table[i].clock;
	block["stdby"] = dim_table[i].stdby;
	block["dist"] = dim_table[i].dist;
	block["ldr"] = dim_table[i].ldr;	
	for (uint8_t j = 0; j < (4 - 1); j++) {				
		JsonObject& step = line.createNestedObject("step");
		step["time"] = dim_table[i].dim_step[j].time;
		step["h"] = dim_table[i].dim_step[j].h;
		step["s"] = dim_table[i].dim_step[j].s;
		step["v"] = dim_table[i].dim_step[j].v;
		step["w"] = dim_table[i].dim_step[j].w;
		step["ww"] = dim_table[i].dim_step[j].ww;
	}
}
```
i,j < (4 - 1) needs to be changed to dynamically adapt to more ore less nodes, if the user will delete some or add some in the WebUI

but i assume, i am supposed to use "add" ?! I am also wondering, if i better should go for the new jsonDocument with v6 of arduinoJson?

In my WebSocket.js on the Client i am parsing the JSON:

```js
// Listen for messages
connection.onerror = function (error) {
    console.log('WebSocket Error ', error);
};
connection.onmessage = function (event) {
	json = JSON.parse(event.data);
	console.log('Server: ', json);
};
```
which was initially sent from the ESP:
```c
webSocket.sendTXT(client, json.c_str(), json.length());
```
From the JSON received the browser UI on the client draws a Line chart with draggable circles using D3js:
https://bl.ocks.org/denisemauldin/538bfab8378ac9c3a32187b4d7aed2c2

Currently, my LineGraph.js only works with a simple hard coded JSON to show the draggable line chart:

```js
let points = d3.range(0, 4).map(function(i) {	
	try {
			console.log('D3: ', json[i].x,json[i].y);			
			return [json[i].x,json[i].y];
		} catch (e) {
			console.log('No Data');
		}
		});
```

Any comments are welcome. thanks!
regards, schmonet

_Originally posted by @schmonet in https://github.com/bblanchon/ArduinoJson/issues/825#issuecomment-454801653_

bblanchon, 2019-01-19:
Hi @schmonet,

You create an array at the root with `createArray()`, but then you treat it like an object with `createNestedObject("block")`.
That's why you get the following error:

    error: no matching function for call to 'ArduinoJson::JsonArray::createNestedObject(const char [6])'

(assuming that's what you meant by "it wont work").

Look at your document, there is an object in the root array, so you must first call `root.createNestedObject()` and then add the values to this object.

Regards,
Benoit

schmonet, 2019-01-21:
Hi Benoit,
ok, thanks! I will it today evening.
Regards, Martin.

schmonet, 2019-01-22:
Ok, ich changed
```c
JsonArray& root = jsonBuffer.createArray();
```
to 
```c
JsonObject& root = jsonBuffer.createObject();
JsonObject& line = jsonBuffer.createObject();
```
now it compiles with no errors. I need to check the JSON-documend later, if it looks like it should.

Thanks!

schmonet, 2019-01-25:
Now i have the following:
```c
for (uint8_t i = 0; i < 4; i++) {
	JsonObject& block = root.createNestedObject();
	block["clock"] = dim_table[i].clock;
	block["stdby"] = dim_table[i].stdby;
	block["dist"] = dim_table[i].dist;
	block["ldr"] = dim_table[i].ldr;
      JsonArray& steps = **root**.createNestedArray();
      for (uint8_t j = 0; j < 4; j++) {       
        JsonObject& step = steps.createNestedObject();
        step["time"] = dim_table[i].dim_step[j].time;
        step["h"] = dim_table[i].dim_step[j].h;
        step["s"] = dim_table[i].dim_step[j].s;
        step["v"] = dim_table[i].dim_step[j].v;
        step["w"] = dim_table[i].dim_step[j].w;
        step["ww"] = dim_table[i].dim_step[j].ww;

      }
}
```
which results in:
```json
[{"clock":480,"stdby":20,"dist":0,"ldr":50},[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":3,"h":0,"s":0,"v":500,"w":0,"ww":0},{"time":30,"h":0,"s":0,"v":50,"w":0,"ww":0},{"time":30,"h":0,"s":0,"v":0,"w":0,"ww":0}],{"clock":720,"stdby":0,"dist":1,"ldr":100},[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":1,"h":0,"s":0,"v":100,"w":0,"ww":0},{"time":20,"h":0,"s":0,"v":100,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":0,"w":0,"ww":0}],{"clock":1080,"stdby":0,"dist":1,"ldr":50},[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":3,"h":0,"s":0,"v":30,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":30,"w":0,"ww":0},{"time":300,"h":0,"s":0,"v":0,"w":0,"ww":0}],{"clock":1260,"stdby":1,"dist":20,"ldr":0},[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":6,"h":0,"s":0,"v":10,"w":0,"ww":0},{"time":60,"h":0,"s":0,"v":10,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":0,"w":0,"ww":0}]]
```
this is not yet, what i want, because i want:
```json
[{"clock":480,"stdby":20,"dist":0,"ldr":50,"step":[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":3,"h":0,"s":0,"v":500,"w":0,"ww":0},{"time":3,"h":0,"s":0,"v":50,"w":0,"ww":0},{"time":30,"h":0,"s":0,"v":0,"w":0,"ww":0}]},{"clock":720,"stdby":0,"dist":1,"ldr":100,"step":[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":1,"h":0,"s":0,"v":100,"w":0,"ww":0},{"time":20,"h":0,"s":0,"v":100,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":0,"w":0,"ww":0}]},{"clock":1080,"stdby":0,"dist":1,"ldr":50,"step":[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":3,"h":0,"s":0,"v":30,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":30,"w":0,"ww":0},{"time":300,"h":0,"s":0,"v":0,"w":0,"ww":0}]},{"clock":1260,"stdby":1,"dist":20,"ldr":0,"step":[{"time":0,"h":0,"s":0,"v":0,"w":0,"ww":0},{"time":6,"h":0,"s":0,"v":10,"w":0,"ww":0},{"time":60,"h":0,"s":0,"v":10,"w":0,"ww":0},{"time":10,"h":0,"s":0,"v":0,"w":0,"ww":0}]}]
```
when i change the code above to
```c
for (uint8_t i = 0; i < 4; i++) {
	JsonObject& block = root.createNestedObject();
	block["clock"] = dim_table[i].clock;
	block["stdby"] = dim_table[i].stdby;
	block["dist"] = dim_table[i].dist;
	block["ldr"] = dim_table[i].ldr;
      JsonArray& steps = **block**.createNestedArray();
      for (uint8_t j = 0; j < 4; j++) {       
        JsonObject& step = steps.createNestedObject();
        step["time"] = dim_table[i].dim_step[j].time;
        step["h"] = dim_table[i].dim_step[j].h;
        step["s"] = dim_table[i].dim_step[j].s;
        step["v"] = dim_table[i].dim_step[j].v;
        step["w"] = dim_table[i].dim_step[j].w;
        step["ww"] = dim_table[i].dim_step[j].ww;

      }
}
```

i get the error:
note:   candidate expects 1 argument, 0 provided
       JsonArray& steps = block.createNestedArray();
                                                  ^
exit status 1
no matching function for call to 'ArduinoJson::JsonObject::createNestedArray()'

Can you help? Thanks!

schmonet, 2019-01-25:
while just writing the question i found the answer ;-)
I needed to add "steps" to identify the nestedArray with a name:
```c
JsonArray& steps = block.createNestedArray("steps");
```

bblanchon, 2019-01-27:
Congratulations!
I'm glad you manage to get your way around.

Thank you for the update, and thank you for using ArduinoJson!

closed 2019-01-27 by bblanchon