bblanchon/ArduinoJson #900: Something wrong with <int> default value in ArduinoJSON v.6.7.0 beta

Warlib1975, 2019-01-29:


Warlib1975, 2019-01-29:
Hello,
the default value works not properly in ArduinoJSON v.6.7.0 beta. 
 
slaveItem.RxPin 		= slave["RxPin"].as<int>() | -1;
slaveItem.TxPin 		= slave["TxPin"].as<int>() | -1;

here is part of JSON
      "RxPin": 15,
      "TxPin": 13,

After parsing, RxPin and TxPin is equal -1, insted of 15, 13, correspondingly.  

If I remove "| -1" 
slaveItem.RxPin 		= slave["RxPin"].as<int>();
slaveItem.TxPin 		= slave["TxPin"].as<int>();
everything is ok. The code returns 15, 13. 

Platfirm: ESP8266.
ArduinoJSON: v.6.7.0 beta

bblanchon, 2019-01-29:
Hello @Warlib1975,

To use the [operator|](https://arduinojson.org/v6/api/jsonvariant/or/?utm_source=github&utm_medium=issues), you must not call [JsonVariant::as<T>()](https://arduinojson.org/v6/api/jsonvariant/as/?utm_source=github&utm_medium=issues):

```c++
slaveItem.RxPin = slave["RxPin"] | -1;
slaveItem.TxPin = slave["TxPin"] | -1;
```

Regards,
Benoit

closed 2019-02-18 by bblanchon

Warlib1975, 2019-02-18:
Hello, 

thank you. It works. Excuse me for the question.

bblanchon, 2019-02-18:
No problem @Warlib1975. Thank you for using ArduinoJson.