bblanchon/ArduinoJson #901: Error compiling project for AArduino Due

VadimCheg, 2019-01-30:
HI. Have got small trouble. Prepared sketch for Arduino Leonardo with Ethernet shield. Am using Arduino IDE 1.8.8, ArduinoJson 5.13.4. It compiles and works at Leo. Now I want to upload that sketch to Arduino Due and get compilation error (no changes were applied to sketch):
C:\Users\...\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp:50:35: error: invalid conversion from 'ArduinoJson::Internals::JsonVariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]

     return impl()->template as<T>();

Hope this can be solved easily.
Thanks in advance!

bblanchon, 2019-01-31:
Hi @VadimCheg,

Please check out [Error: invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v5/faq/invalid-conversion-from-const-char-to-char/?utm_source=github&utm_medium=issues)

Regards,
Benoit

VadimCheg, 2019-02-02:
Hello Benoit!

Thank You very much! Sorry for reporting problem which is actually in my code. I was confused because it perfectly compiled for Uno.

Thanks again!

Best regards,
Vadim.

closed 2019-02-02 by VadimCheg

bblanchon, 2019-02-03:
Hi Vadim,

> it perfectly compiled

Is that so?
Or did you overlook compiler warnings :wink:

Thank you for using ArduinoJson.
Don't forget to star the project on GitHub if you like it!

Regards,
Benoit