bblanchon/ArduinoJson #906: serialize to yun sd by fileIO.h

novski, 2019-02-13:
i found a lot of parsing help but none related to serialize.

I have some current Sensors writing a nice Json to Serial but i can't save it to a file on my yun's SD card...

How can i serialize a json file and write it to my yun sd card?

This is my sketch:
```
#include <Bridge.h>
#include <FileIO.h>
#include <ArduinoJson.h>
#include "Arduino.h"
#include <INA.h> // Zanshin INA Library

const uint32_t SERIAL_SPEED = 115200;  ///< Use fast serial speed
uint8_t        devicesFound = 0;       ///< Number of INAs found
INA_Class      INA;                    ///< INA class instantiation

void setup() {
  devicesFound = INA.begin(1,100000); // Set to an expected 10 Amp maximum and a 100000 microOhm resistor
  INA.setBusConversion(8500);            // Maximum conversion time 8.244ms
  INA.setShuntConversion(8500);          // Maximum conversion time 8.244ms
  INA.setAveraging(128);                 // Average each reading n-times
  INA.setMode(INA_MODE_CONTINUOUS_BOTH); // Bus/shunt measured continuously
  INA.AlertOnBusOverVoltage(true,5000);  // Trigger alert if over 5V on bus
  // a filesystem handler throug bridge.h
  FileSystem.begin();
} // method setup()

void loop() {
  StaticJsonBuffer<512> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  for (uint8_t i=0;i<devicesFound;i++){ // Loop through all devices
    String CurrentSensorNr = "CurrentSensor";
    CurrentSensorNr += i+1;
    JsonArray &CurrentSensor = root.createNestedArray(CurrentSensorNr);
      JsonObject &CurrentSensorObj = CurrentSensor.createNestedObject();
                  CurrentSensorObj["DeviceAddress"] = INA.getDeviceAddress(i);
                  CurrentSensorObj["DeviceName"] = INA.getDeviceName(i);
                  CurrentSensorObj["BusMilliVolts"] = INA.getBusMilliVolts(i) / 1000.0;
                  CurrentSensorObj["ShuntMicroVolts"] = INA.getShuntMicroVolts(i) / 1000.0;
                  CurrentSensorObj["BusMicroAmps"] = INA.getBusMicroAmps(i) / 1000.0;
                  CurrentSensorObj["BusMicroWatts"] = INA.getBusMicroWatts(i) / 1000.0;
  } // for-next each INA device loop
  SerialUSB.println("\n");
  root.prettyPrintTo(Serial); // rettyP
  File sensorFile = FileSystem.open("/tmp/sensors.json", FILE_WRITE);
  root.prettyPrintTo(sensorFile);
  sensorFile.close();
  delay(5000); // Wait 5 seconds before next reading
} 
```
this is what it prints to serial:
```
20:36:57.307 -> {
20:36:57.307 ->   "CurrentSensor1": [
20:36:57.343 ->     {
20:36:57.343 ->       "DeviceAddress": 65,
20:36:57.343 ->       "DeviceName": "INA219",
20:36:57.343 ->       "BusMilliVolts": 5.008,
20:36:57.343 ->       "ShuntMicroVolts": 41.96,
20:36:57.343 ->       "BusMicroAmps": 419.561,
20:36:57.343 ->       "BusMicroWatts": 2100.859
20:36:57.343 ->     }
20:36:57.343 ->   ],
20:36:57.343 ->   "CurrentSensor2": [
20:36:57.343 ->     {
20:36:57.343 ->       "DeviceAddress": 66,
20:36:57.343 ->       "DeviceName": "INA219",
20:36:57.343 ->       "BusMilliVolts": 5.012,
20:36:57.343 ->       "ShuntMicroVolts": 21.03,
20:36:57.343 ->       "BusMicroAmps": 210.269,
20:36:57.343 ->       "BusMicroWatts": 1052.871
20:36:57.343 ->     }
20:36:57.343 ->   ],
20:36:57.343 ->   "CurrentSensor3": [
20:36:57.343 ->     {
20:36:57.343 ->       "DeviceAddress": 67,
20:36:57.343 ->       "DeviceName": "INA219",
20:36:57.343 ->       "BusMilliVolts": 4.992,
20:36:57.343 ->       "ShuntMicroVolts": 45.24,
20:36:57.343 ->       "BusMicroAmps": 452.337,
20:36:57.343 ->       "BusMicroWatts": 2258.332
20:36:57.343 ->     }
20:36:57.343 ->   ],
20:36:57.343 ->   "CurrentSensor4": [
20:36:57.343 ->     {
20:36:57.343 ->       "DeviceAddress": 68,
20:36:57.343 ->       "DeviceName": "INA219",
20:36:57.343 ->       "BusMilliVolts": 4.988,
20:36:57.343 ->       "ShuntMicroVolts": 36.97,
20:36:57.343 ->       "BusMicroAmps": 369.664,
20:36:57.343 ->       "BusMicroWatts": 1843.897
20:36:57.343 ->     }
20:36:57.343 ->   ]
 }
```

bblanchon, 2019-02-15:
Hi @novski,

I don't have any experience with Arduino Yùn, but it looks like your doing the right thing.
As far as ArduinoJson is concerned, your code is correct.

I'm pretty sure that your problem is not related to ArduinoJson.
Are you able to create a file on the SD card without the help of ArduinoJson?

However, I notice that you didn't call `Bridge.begin()`, could that be the problem?

Regards,
Benoit

novski, 2019-02-16:
It was exactly that! Forgive me. Thanks a lot!

bblanchon, 2019-02-17:
You're welcome @novski. Thanks for using ArduinoJson.
Don't forget to star the project to spread the word 😉

closed 2019-02-17 by bblanchon